evcc-io/docs #10: phases Setting

andig, 2021-11-18:
Ist nur eine Rechenhilfe, stellt aber keine Phasen ein

DerAndereAndi, 2021-11-18:
- Sollten wir dann in der evcc.dist.yaml auch so erklären, da denkt jeder dass mit dem setting etwas fix eingestellt wird
- Sollte dann auch im docs repo so erklärt werden.

Wie wäre es mit:

```
phases: 3 # ev phases (default 3), only needed and relevant if the charger has no meter (assigned)
```

andig, 2022-05-05:
@RTTTC do you want to make a suggestion for the docs? We do not have the capacity to handle the bumps...

VolkerK62, 2022-05-26:
@andig  Bei phases hat sich ja seit Nov einiges getan, ist diese issue eigentlich noch aktuell? 
Wenn ja, hätte ich ein paar technische Fragen, bevor ich wieder "Mist" schreibe ;-)

andig, 2022-05-26:
Frag ;)

VolkerK62, 2022-05-27:
Ich denke, man sollte erwähnen, dass dieser Parameter nichts am physikalischen Anschluß ändert, sondern lediglich ein Berechnungswert ist ;-)

1. Solange noch nicht geladen wird, wird `phases` dazu verwendet, um aus dem verfügbaren PV-Überschuss einen Ladestrom je Phase zu ermitteln? 
2. Wenn sowohl bei `vehicle` als auch bei `loadpoint" ein `phases` Wert eingetragen ist, wird dann der kleinere der beiden für die Berechnung genommen? Bei unbekanntem Fahrzeug also immer `loadpoint`.
3. Wenn geladen wird, werden (wenn die Wallbox bzw. der Ladezähler das unterstützt) die Phasen aufgund der aktuellen Ladeströme ermittelt? Der Wert aus der yaml wird dadurch overruled.? Wie lange wird dieser aktuelle Wert beibehalten? Solange das Fahrzeug angesteckt ist oder solange der Ladevorgang läuft, oder ...?
4. Es gibt Fälle, das wird von 2 Fahrzeugen eines nicht erkannt, weil offline. Könnte man in Bezug auf Pkt 3 den Trick anwenden, dass man die Ladung mit "MinPV" startet, die tatsächlcihen Phasen werden erkannt und dann stellt man nach kurzer Zeit auf "PV" um und alles funzt wunderbar?

andig, 2022-05-27:
> Ich denke, man sollte erwähnen, dass dieser Parameter nichts am physikalischen Anschluß ändert, sondern lediglich ein Berechnungswert ist ;-)

Absolut. Und dass sie bei schaltbaren Chargern ignoriert wird.

> 1. Solange noch nicht geladen wird, wird `phases` dazu verwendet, um aus dem verfügbaren PV-Überschuss einen Ladestrom je Phase zu ermitteln?

Es wird verwendet, um zu prüfen ob die minimale Startleistung (!) erreicht ist. Die hängt ja von den Phasen hab- nämlich dem Minimum aus Loadpoint und erkanntem Fahrzeug!

> 2. Wenn sowohl bei `vehicle` als auch bei `loadpoint` ein `phases` Wert eingetragen ist, wird dann der kleinere der beiden für die Berechnung genommen? Bei unbekanntem Fahrzeug also immer `loadpoint`.

Check.

> 3. Wenn geladen wird, werden (wenn die Wallbox bzw. der Ladezähler das unterstützt) die Phasen aufgrund der aktuellen Ladeströme ermittelt?

Ja. Da kann natürlich nur passieren wenn LP=3 konfiguriert ist und das Fahrzeug weniger nimmt.

> Der Wert aus der yaml wird dadurch overruled.? 

Der Wert ist nur eine Rechenhilfe solange nix Besseres bekannt ist. "Besseres" ist 

a) ein erkanntes Fahrzeug mit Phasenkonfiguration oder
b) eine Messung der Ladeströme

> Wie lange wird dieser aktuelle Wert beibehalten? Solange das Fahrzeug angesteckt ist oder solange der Ladevorgang läuft, oder ...?

Logischerweise solange Fahrzeug dran- das ändert sich ja nicht mehr.

> 4. Es gibt Fälle, das wird von 2 Fahrzeugen eines nicht erkannt, weil offline. Könnte man in Bezug auf Pkt 3 den Trick anwenden, dass man die Ladung mit "MinPV" startet, die tatsächlcihen Phasen werden erkannt und dann stellt man nach kurzer Zeit auf "PV" um und alles funzt wunderbar?

Wenn die Ströme gemessen werden ja.

Jetzt bin ich gespannt, wie Du das in eine konsistente Story bringst ;)

VolkerK62, 2022-05-27:
> Absolut. Und dass sie bei schaltbaren Chargern ignoriert wird.

Nicht ganz. Bei denen muss `phases` zwingend auf `1` stehen.

> Es wird verwendet, um zu prüfen ob die minimale Startleistung (!) erreicht ist. Die hängt ja von den Phasen hab- nämlich dem Minimum aus Loadpoint und erkanntem Fahrzeug!

Also `minCurrent` * `(min)phases` *  230V = minimale Startleistung

> Jetzt bin ich gespannt, wie Du das in eine konsistente Story bringst ;)

Du baust ja ganz schön Druck auf ;-)

VolkerK62, 2022-05-27:
Da ich mir mit Änderungen im PR-Prozedere noch schwer tue, hier mal ein Text-Entwurf:

Dieser Parameter ändert nichts am physikalischen Anschluss der Wallbox, sondern dient lediglich dazu, (in Verbindung mit „loadpoint minCurrent“) die minimale Startleistung für die Ladung zu ermitteln.
Dafür wird der kleiner Wert aus „vehicle phases“ und „loadpoint phases“ zur Berechnung herangezogen. Bei unbekannten Fahrzeugen wirkt also immer „loadpoint phases“.
Wenn die Ladung läuft und die Wallbox bzw. der Ladezähler die Phasenströme liefert, wird daran die tatsächliche Anzahl der Phasen erkannt und (solange das Fahrzeug angesteckt bleibt) für die weitere Berechnung der Ladeleistung genutzt.

VolkerK62, 2022-05-27:
Habs einfach mal gewagt ;-)   PR ist raus #130

VolkerK62, 2022-05-28:
Kann geschlossen werden. PR ist merged.

andig, 2022-05-28:
Setz mal künftig in Deine PRs ein

    Fix #xyz

dann werden die Issues automatisch geschlossen!

closed 2022-05-28 by andig

VolkerK62, 2022-05-28:
ok, ich hab ja noch einen offen. Aber wo setze ich das ein?  Als Kommentar?

VolkerK62, 2022-05-28:
Hat sich (glaube ich) erledigt. Wenn ich das mit meinem 45 Jahre alten Schulenglisch richtig verstanden habe muss man das beim Erstellen der PR in die Beschreibung einfügen. nachträglich geht es wohl  nicht mehr.