evcc-io/docs #16: Konsistente Einrückung

andig, 2021-12-05:
Bisher hatten wir

```yaml
foo:
- bar
```

Als Einrückung durchgängig verfolgt. Unter [Go-](https://docs.evcc.io/docs/devices/chargers#go-echarger-homeprov3) sehe ich 

```yaml
chargers:
  - name: my_charger
    type: go-e
    token: 4711c # go-e cloud API token
    cache: 10s # go-e cloud API cache duration
```

Beides ist richtig, Konsistenz wäre dennoch schön.

andig, 2021-12-07:
Prettier (wie auch das Redhat YAML Plugin) haben sich entschlossen, immer einzurücken: https://github.com/prettier/prettier/issues/4723

DerAndereAndi, 2021-12-08:
Der Inhalt kommt vom config repo. Also sollte es dort korrigiert werden, falls nötig.

naltatis, 2022-03-26:
@andig Soll ich das mal in einem Commit komplett gerade ziehen und überall auf die prettier-Art formatieren?

andig, 2022-03-27:
Für mich ist nicht erkennbar, was besser/schlechter ist und wo dann wieder vscode von Hand konfiguriert werden will. Vmtl. tun wir uns einen Gefallen, wenn wir dem De-facto Standard Prettier folgen?

> @andig Soll ich das mal in einem Commit komplett gerade ziehen und überall auf die prettier-Art formatieren?

Ich wär dafür, fände es aber gut das vorab als separate PRs für core und docs zu machen.

closed 2022-04-06 by naltatis