evcc-io/docs #17: Übersichtskapitel ist leer

andig, 2021-12-05:
https://docs.evcc.io/docs/guides/overview kann entfallen?

DerAndereAndi, 2021-12-08:
Und was wäre dann der Einstieg? Direkt in ein Thema rein finde ich nicht gut. Die Frage für mich wäre eher: was für einen Inhalt sollte man da anzeigen

andig, 2021-12-20:
Also so wie es jetzt ist kann die Seite auch einfach weg da überhaupt keine zusätzlichen Informationen enthalten sind.

VolkerK62, 2022-05-27:
Ich hätte einen Vorschlag zur Umstrukturierung des gesamten Kapitel. Allerdings traue ich mich mangels Kenntnis nicht an diesen Umbau. Und an das Menü am linken Rand komme ich auch nicht ran.

Den Punkt "Anleitungen" umbenennen in "Häuftige Fragen und Themen".
Darin dann die einzelnen Unterpunkte
- Einrichtung und Konfiguration
- Laden
- Wallboxen
- Hausinstallation
- Fahrzeuge
- best practice (Tipps und Tricks)
- deep inside (wie rechnet und steuert evcc)
  
Beim aktuellen "Konfiguration" ist nur die Ersparnisanzeige beschrieben, Das könnte man zum neuen Unterpunkt "Laden" dazupacken oder zu "best practice"

Bei aktuellen "Laden" steht ein Tip zum laden im Winter drin. Den könnte man auch zum neuen Laden packen oder zu "Tbest practice" . 
Diese Rubrik könnte man dann auch in Zukunft immer weiter fortschreiben. Gefüttert durch Fragen und Antworten aus den Diskussionen.

Unter dem Punkt "deep iside" könnte man dann (etwas feiner, aber nicht bis ins kleineste Detail) die Berechnungs- und Steuerungslogik beschreiben. Das hilft evtl den Anwendern, wenn mal etwas nicht so richtig klappt. 

Inhalte kann ich gerne beisteuern, habe aber mit dem Umbau des Menüs und mit den Formatierungen so meine Probleme.

VolkerK62, 2022-05-28:
@andig  Ich habe mich mal rangetraut und die Unterpunkte im Block "Anleitungen" mit den jeweiligen Inhalten aus "Häufige Fragen und Antworten" gefüllt. Diesen Punkt habe ich dann umbenannt in "Wallboxxen und Hausinstallation". Das war der Rest der noch übrig geblieben ist. Evtl sollte man da 2 Punkte draus machen, aber da wusste ich nciht, wie das geht.

PR #132 (2022-06-03): Update setup.mdx
fix #17
PR #133 (2022-06-03): Update charging.mdx
fix #17
PR #134 (2022-06-03): Update vehicles.mdx
fix #17
PR #135 (2022-06-03): Update faq.mdx
fix #17
closed 2022-06-03 by premultiply