evcc-io/docs #28: Wallbe Dokumentations Update erbeten

Hofyyy, 2021-12-29:
Seit Version 0.75 und neuen Wallbe Firmware Versionen ist der Eintrag NOTE soweit ich das sehe überholt.

Vielleicht könnte man hier folgenden schreiben:

Für eine bestmögliche Kompatibilität der Modbus TCP implementation wird
mindestens folgende Firmware empfohlen.
- Wallbe Eco 2.0 - SL-01.04.21
- Wallbe Pro - BE-03.01.04

Die Firmware Versionen können hier angefordert werden:
https://www.wallbe.de/firmware-updates/

andig, 2021-12-29:
@premultiply ?

premultiply, 2021-12-29:
Jaha. Morgen...

Hofyyy, 2021-12-30:
Ich weiss nicht wie ihr das sehr, aber man sollte die Firmware zumindest offiziell nicht weitervertreilen und in der Doku lieber auf die Firma verweisen oder? Ich meine sonst muss man ja gefühlt die Box "supporten"...

andig, 2021-12-30:
Muss ja niemand der keinen Sdm verwenden will. Das sollte klar werden.

Hofyyy, 2021-12-30:
Die Doku liest sich ausgezeichnet.

Letzte Frage. Soll das legacy im config Beispiel noch drin bleiben auf der Dokuseite?

premultiply, 2021-12-30:
Oh, das hab ich vor lauter Text glatt vergessen! Danke!

Hofyyy, 2021-12-30:
Wenn ich schon noch keine Sourcecodeänderungen einbringe. So lese ich mindestens schon mal alles mit und gegen :-)

premultiply, 2021-12-30:
Ich hoffe nur die Templates und das Frontend aktualisieren sich nun irgendwann richtig und ich habe nichts falsch gemacht... :)

closed 2022-01-03 by DerAndereAndi

horstvogel, 2023-07-20:
Moin, jemand eine Idee wo ich für unsere Wallbe Pro mit EV Chargre Control RCM einen Firmware Download erhalte. Von Wallbe Compleo habe ich bisher noch keine Rückmeldung erhalten. Wir sind noch auf der Firmware Version 0,72, also sehr, sehr alt. Der Controler ist doch eigentlich von Phonenix? 
Vielen Dank!

Hofyyy, 2023-07-20:
Die musst du hier per Email anfragen: mailto:wallbe-firmware@compleo-cs.com

horstvogel, 2023-08-01:
Kurzes Update, Wallbe Compleo hat sich nicht gemeldet. Ich hätte das auch ehrlich gesagt nicht wirklich erwartet....anscheinend ist jetzt mit der Wallbox auch halbwegs ruhe, Wenn jetzt ein Update kommt von EVCC, dann starte ich nach dem Update einfach auch die Wallbox über die Weboberfläche vom Wallbox Controller neu. Vielleicht war das die "Lösung" 
Danke!

premultiply, 2023-08-01:
Du kannst die Original-FW bei Phoenix einfach runterladen.
Welchen Controller hast du denn genau.
Mach mal bitte ein Bild von Innenleben deiner Box.

Ansonsten ist ja oben eine aktuellere Wallbe-Firmware für dich verlinkt.

horstvogel, 2023-08-01:
Vielen, Dank für die Hilfe. 

Das ist die Wallbe pro Box. Die Zähler sind eine Besonderheit, die leider auch eher unschön sind. Wenn jemand einen Tipp hat, welche Original Modbus Zähler da 1zu1 gleich funktionieren würden, dann würde ich die auch gerne ersetzen. 

[image]

premultiply, 2023-08-01:
Das sind die ganz normalen Phoenix-Controller wie sie auch in der Eco 2.0s drin sind. Nur halt mit uralter Firmware.
Das sind diese hier: https://www.phoenixcontact.com/de-de/produkte/ac-ladesteuerung-ev-cc-ac1-m3-cbc-rcm-eth-1018701
Dort gibt es auch die aktuelle Phoenix-Firmware. (Kann keine 0,1A-Schritte, nur 1A-Schritte)

Für die aktuelle Wallbe-Firmware (mit 0,1A-Schritten) melde dich mal im Slack-Kanal bei mir.

Wichtig: Vor dem Update die IP-Adresse immer auf Standard (zurück)stellen und am besten auch direkt mit dem Rechner dran.
Andernfalls ist das Update sehr hakelig weil sich zwischendrin die IP umstellt.

horstvogel, 2023-08-07:
Nur zur Info:
die Firmware mit der Versionsnummer  SL.01,04,24 sind nur auf der Wallbe Pro drauf, leider ist 2 Tagen die Box wieder nicht erreichbar, weder über EVCC hier jetzt mit der Fehlermeldung EOF, vorher war die Meldung "charger: dial tcp ...connect: no route to host" 
und auch nicht über das Webinterface ist die Box wieder nicht erreichbar. 

Somit war die "alte" Firmware dann doch leider nicht das Problem. 

Muss ich mal schauen, was da dann das Problem vielleicht ist....

Vielen Dank nochmals für die Hilfe die man hier wirklich immer erhält!

Hofyyy, 2023-08-07:
Hallo,

mal eine ganz doofe Idee. Hast du schon mal ein anderes Netzwerkkabel probiert, oder einen anderen Port? Eine statische IP?

Also ich habe auch die Wallbe Pro laufen mit der verlinkten Firmware. Und sie läuft wirklich komplett stabil.

daher die frage nach der Infrastruktur ...

horstvogel, 2023-08-07:
Statische IP, 2 Controller auf 2 unterschiedlichen Ports...beide Wallboxen sind mal irgendwann nicht erreichbar. Kann mir noch keine Reim draus machen, warum.

horstvogel, 2023-08-08:
Moin, gestern dann versucht die Wallbox über ein direkt an die Wallbox angeschlossenes Kabel zureichen. 
Einmal mit meiner jetzt eingestellten festen IP, also den PC auch in gleich Subnet.....dann einmal über die Wallbe default IP, PC dann auch ins default Subnet.... 
Auf beiden Wegen war die Wallbox nicht erreichbar. Somit würde ich vorsichtig das Netzwerk ausschließen und die Ladekontroller oder die Konfiguration eher  Problem Ursache in Verdacht haben. 

Dann einmal den Resetknopf kurz gedrückt, danach war die Wallbox wieder erreichbar, sowohl via Webinterface als auch über Modbus durch EVCC. Strom aus und Strom an sorgte nicht dafür, dass die Wallbe wieder erreichbar war. Das finde ich dann ganz seltsam

Habe dann einmal einen Werksresett durchgeführt, die aktuelle Firmware nochmals eingespielt und dann wieder ins Netzwerk via fester IP ...gebracht. Mal schauen, ob das dann jetzt läuft. Das erste Firmware Update auf SL-01.04.24 hatte geklappt, ohne das ich die default IP 192.168.0.8 gesetzt hatte, dieses wurde ja auch hier empfohlen. Mal schauen was dann jetzt so passiert.

Status der Wallbe Pro ohne das ein Auto angeschlossen ist. **Hier die Frage, Status ist wirklich Status? Oder kann/sollte man hier auch noch was einstellen??**

[image]

Die Konfiguration der Wallbe Pro, wie Sie bei mir eingestellt ist. Die Box hat **keinen** Zähler der via Modbus mit dem Controller kommuniziert. Der Zähler dient lediglich der Energieerfassung und hatte früher die Freigabe der Wallbox über den SmartMe Zähler über die Cloud der Firma eCarUp der Wallbox via potentialfreier Kontakte gesteuert. Die Zähler fliegen jetzt aber demnächst raus. 

[image]

Danke!

Hofyyy, 2023-08-08:
Hallo, dann drücken wir mal die Daumen. Verwirrt bin ich bei deiner Firmware. Bei mir steht bei der Wallbe Pro diese hier: Version: BE-03.01.29

horstvogel, 2023-08-08:
> Hallo, dann drücken wir mal die Daumen. Verwirrt bin ich bei deiner Firmware. Bei mir steht bei der Wallbe Pro diese hier: Version: BE-03.01.29

Verwirrt bin ich zwischenzeitlich auch schon ;-) die BE-03.01.29 frisst die Wallbe Pro nicht, Sie bleibt immer im Firmware Uploadbereich hängen. mit der "falschen Firmware" SL-01.04.24   funktioniert das. Das einzige was noch nicht probiert wurde, ich das BE-xxxx Update mit dem Internet Explorer als Browser durchzuführen. Zum Glück hat ein Experte auf dem Notebook Windows 11 installiert, da gibt es dann kein Internet Explorer mehr drauf

Vielen Dank das Ihr mich hier unterstützt.

csigristgit, 2023-11-30:
Ich habe aktuell dieselbe Problematik. Hat das mit dem Update auf die Version BE-xxx geklappt? Das ganze ist schon echt mühsam aufgebaut. Aktuell habe ich die 0.88, was ebenfalls Steinalt ist...

Grüsse

daniell82, 2024-08-12:
Ich bin auf der Suche nach einer Wallbe Pro Firmware, Compleo habe ich angeschrieben auch wenn ich mir da keine Hoffnungen machen. Hat die Firmware einer von euch und kann mir die zusenden? Wäre echt mega, hab grad Probleme mit der offiziellen Firmware 2.27 und der Anbindung des Modbus Zählers...

vannilla1, 2024-09-12:
Hello.

Hat jemand Erfahrung mit wallbe Pro Plus in Verbindung mit der eCarUp-Software? Ich würde gerne wissen, ob Sie Probleme haben, welche Firmware Sie haben usw., denn es gibt niemanden mehr von Wallbe oder eCarUp, der uns dabei helfen kann. Wir haben derzeit 46 defekte Punkte, die nicht mit der eCarUp-Software kommunizieren und selbst nach dem Anschließen der Station an den PC und dem Zurücksetzen wird die Verbindung nicht wiederhergestellt. Ich bin verzweifelt.

naltatis, 2024-09-13:
@vannilla1 das hier ist unser Doku Repository. Würde dir empfehlen, die Frage in unseren Discussions im Hauptprojekt noch mal zu stellen. Da lesen viele Leute mit. https://github.com/evcc-io/evcc/discussions

KoMa1978, 2025-04-23:
> kurz gedrückt, danach war die Wallbox wieder erreichbar, sowohl via Webinterface als auch über Modbus durch EVCC. Strom aus und Strom an sorgte nicht dafür, dass die Wallbe wieder

Hi,

passt vielleicht nicht ganz in den Thread, aber die Verbindungsprobleme kenne ich auch (mit der ECO 2.0s) ...
Das Problem tritt immer mal wieder auf ... eher unregelmäßig, gefühlt öfter, je wärmer es draußen wird.
Mach ich kurz den FI kurz aus und an, ist erstmal wieder alles OK
Das Problem kann man reproduzieren, wenn man den Deckel abnimmt und das gesteckte Netzwerkkabel bzw die Buchse in eine Richtung drückt.

Ich habe den LAN Port (Würth Electronik 74990101210) schonmal nachgelötet, bisher ohne Erfolg. Bin schon am überlegen, mir mal eine neue RJ45 Buchse zu bestellen. Ich vermute, der RJ45 Port hat eine Macke oder die Platine. Ich bin mir auch nicht sicher, ob da ein Magnetkontakt im Port ist, laut Beschreibung steht da was von Magnet und ohne Kabel kann ich mit dem Multimeter keinen Kontakt zwischen Lötstelle und Pin in der Buchse messen.

Hofyyy, 2025-10-17:
Bin gerade auf Reise ...

kuehnelfred-cyber, 2026-05-03:
Guten Abend, ich versuche 2 Wallbe Pro wieder fit zu machen. Die Firmwares die ich bisher probiert habe geben keine Ladewerte aus. Könntet ihr mir bitte die BF-01.04.24. schicken? Vielen Dank. @vannilla1 @premultiply @wiinf

vannilla1, 2026-05-03:
> Guten Abend, ich versuche 2 Wallbe Pro wieder fit zu machen. Die Firmwares die ich bisher probiert habe geben keine Ladewerte aus. Könntet ihr mir bitte die BF-01.04.24. schicken? Vielen Dank. @vannilla1 @premultiply @wiinf 

Hallo. Ist die Firmware in ihrer aktuellen Form verfügbar?

commit 318b12d7 (2021-12-30): Update Wallbe

Fixes #28