evcc-io/docs #33: log, levels

patbab, 2022-01-07:
Ich habe gerade mit den log-Einstellungen gespielt.
In der Doku steht bei den levels:

`fahrzeugname: Jedes definierte vehicle (Fahrzeug), hier ist der jeweilige Wert des Parameters name anzugeben`

Ist das korrekt, Parameter `name`?

Ich habe folgendes vehicle:

```
vehicles:
- name: 500e
  type: fiat
  title: XXX # display name for UI
  capacity: 42 # kWh
  # [...]
```

Wenn ich bei `levels:` den `name` einegebe, also hier z.B.:
`  500e: trace`
so passiert nichts.

Gebe ich jedoch den `type` ein, also z.B. 
`  fiat: trace`
dann funktioniert es.

logs für das Fahrzeug auf einem niedrigeren level als trace gibt es wohl nicht?

Auch im log selbst erscheint als "Ursprung" des Eintrags bei `log: trace` immer [fiat   ], und nicht [500e   ].

```
[main  ] INFO 2022/01/07 11:52:24 evcc 0.74 (f6d9d5a)
[main  ] INFO 2022/01/07 11:52:24 using config file E:\evcc\evcc.yaml
[main  ] INFO 2022/01/07 11:52:24 listening at 0.0.0.0:7070
[fiat  ] TRACE 2022/01/07 11:52:24 GET https://loginmyuconnect.fiat.com[...]
[fiat  ] TRACE 2022/01/07 11:52:24 [...]
```

Wenn ich mit meiner Annahme richtig liege:
einerseits OK, so lange es so mit `type `geht.
Es würde aber spätestens dann zu Problemen führen, wenn ich 2 Fiats zu Hause stehen hätte, weil man die dann nicht auseinanderhalten könnte.

Patrick

andig, 2022-01-08:
Die Logs gehen immer auf type, nie auf name

VolkerK62, 2022-05-26:
PR  #129  erstellt

PR #129 (2022-06-03): Update log.md
fix #33