evcc-io/docs #38: Messaging Doku braucht ein update

ExecuteGK, 2022-01-12:
docs/docs/reference/configuration/messaging.md listet einige Variabeln die in Nachrichten verwendet werden können. Allerdings sind die bei weitem nicht komplett und teilweise nicht mehr verfügbar.

Hier die ersten Fehlermeldungen:

[push  ] ERROR 2022/01/12 21:22:07 invalid message template for start: wanted: [loadpoint], got: [minCurrent chargePower vehicleOdometer batteryPower chargeCurrents chargeDuration prioritySoC batteryConfigured homePower sponsor gridPower vehiclePresent pvConfigured chargeConfigured connected maxCurrent currency gridConfigured tariffFeedIn chargeCurrent activePhases enabled savingsSince siteTitle title hasVehicle mode batterySoC minSoC connectedDuration vehicleCapacity vehicleRange targetSoC chargedEnergy charging tariffGrid pvPower vehicleTitle phases]

[push  ] ERROR 2022/01/12 21:22:07 invalid message template for stop: wanted: [energy duration], got: [chargeDuration gridPower batterySoC charging climater tariffGrid chargeCurrent homePower enabled batteryPower vehiclePresent currency chargeRemainingDuration maxCurrent activePhases pvConfigured chargePower title savingsSince vehicleTitle minSoC siteTitle chargeCurrents hasVehicle targetTime vehicleRange gridConfigured connected sponsor targetSoC phases vehicleCapacity chargeConfigured tariffFeedIn connectedDuration prioritySoC batteryConfigured pvPower chargedEnergy minCurrent vehicleSoC vehicleOdometer mode]

[push  ] ERROR 2022/01/12 21:28:24 invalid message template for disconnect: wanted: [sideTitle], got: [activePhases pvConfigured climater title minSoC sponsor currency gridPower connected vehiclePresent hasVehicle vehicleRange tariffFeedIn tariffGrid minCurrent vehicleTitle chargeRemainingDuration chargeCurrents targetSoC homePower phases chargeDuration batteryConfigured charging savingsSince gridConfigured vehicleOdometer mode siteTitle chargeConfigured pvPower enabled targetTime vehicleSoC batterySoC vehicleCapacity connectedDuration chargePower chargeCurrent chargedEnergy maxCurrent prioritySoC batteryPower]

Ansonsten alles in allem eine gute Doku.

Grüße

Sebastian

andig, 2022-01-12:
Es sollte alles gehen was im cache steht

ExecuteGK, 2022-01-12:
Denke ich auch.

Nur zum Anfang ist das immer ein wenig schwierig wenn die Beispiele aus der Doku schon Fehler verursachen.
Vielleicht hat ja jemand sinnvolle Beispiele die schöne Nachrichten produzieren und die Möglichkeiten ein wenig demonstrieren.

Grüße

Sebastian

andig, 2022-01-12:
Bin nicht sicher ob wir das schon haben; es könnte ggf auch spannend sein, Go Templates und Sprig in den Messages verfügbar zu machen. Dann kann der Anwender selbst if/then/else Konstrukte und ähnliches bauen.

ExecuteGK, 2022-01-12:
if/then/else wäre echt super. Dann könnte man sich eine Logik basteln die unter bestimmen Umständen die externe Automatisierung durch eine Nachricht triggert.

Schön wär auch wenn ich durch eine eingehende Nachricht eine Statusübersicht bekomme. Dann muss ich nicht immer erst das VPN starten und die Webseite prüfen wenn ich unterwegs bin. Hab sowas für meine Kameras mit ioBroker und Telegram eingerichtet.

Grüße

Sebastian

DerAndereAndi, 2022-01-13:
Wir freuen uns auch hier über Unterstützung um die Doku besser zu machen :) Am Besten immer von jemandem der eine Funktion auch nutzt. Da ich Messaging nicht nutze, bin ich da nicht so gut geeignet. Und ich muss ja auch nicht alles schreiben :)

ExecuteGK, 2022-01-13:
Wenn es eine Möglichkeit gibt die messages in einem Testmodus alle zu verschicken, dann kann ich die Doku für telegram gerne machen und Beispiele einfügen. Ich will nur nicht zig mal das Auto an und abklemmen müssen. ;)

Grüße 

Sebastian

int5749, 2022-01-14:
Kurze Frage zur bisherigen yaml und der dort enthaltenen Benachrichtigungen.

```
# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
```

Dort sind manche Zeilen komplett in "" andere gar nicht und wieder andere nur zum Teil. Ist dies historisch gewachsen, oder hat dies einen besonderen Grund?

Dies könnte dann bei einem Update der Doku ggfs mit berücksichtigt werden?
if-then-else klingt interessant, nutzte derzeit einige Nachrichten für pushover.

DerAndereAndi, 2022-01-27:
Die Messaging Doku wurde aktualisiert: https://docs.evcc.io/docs/reference/configuration/messaging/

DerAndereAndi, 2022-01-27:
> Dort sind manche Zeilen komplett in "" andere gar nicht und wieder andere nur zum Teil. Ist dies historisch gewachsen, oder hat dies einen besonderen Grund?

Damit das Teil vom gesendeten Text. Also damit werden die Werte einfach etwas hervorgehoben.
 
> Dies könnte dann bei einem Update der Doku ggfs mit berücksichtigt werden? 

Ist imho nicht notwendig.

closed 2022-03-29 by premultiply