evcc-io/docs #41: Add Huawei dongle

andig, 2022-01-15:
Needs connect delay and delay:

    meters:
    - name: grid # Huawei grid power import export (when smart power sensor is present)
      type: custom
      power:
        source: modbus
        timeout: 10s
        connectdelay: 5s
        delay: 2s
        model: huawei
        uri: <ip>:502
        id: 2
        register:
          address: 37113 # Grid import export power
          type: holding
          decode: int32
        scale: -1
      energy:
        source: modbus
        timeout: 10s
        connectdelay: 5s
        delay: 2s
        model: huawei
        uri: <ip>:502
        id: 2
        register:
          address: 37119 # Grid active energy
          type: holding
          decode: int32
        scale: 0.01
      currents:
      - source: modbus
        model: huawei
        uri: <ip>:502 # inverter port
        id: 2
        register:
          address: 37107 # Huawei phase A grid current
          type: holding
          decode: int32
        scale: -0.01
      - source: modbus
        model: huawei
        uri: <ip>:502 # inverter port
        id: 2
        register:
          address: 37109 # Huawei phase B grid current
          type: holding
          decode: int32
        scale: -0.01
      - source: modbus
        model: huawei
        uri: <ip>:502 # inverter port
        id: 2
        register:
          address: 37111 # Huawei phase C grid current
          type: holding
          decode: int32
        scale: -0.01
    - name: battery
      type: custom
      power:
        source: modbus
        timeout: 10s
        connectdelay: 5s
        delay: 2s
        model: huawei
        uri: <ip>:502
        id: 2
        register:
          address: 37001
          type: holding
          decode: int32
      soc:
        source: modbus
        timeout: 10s
        connectdelay: 5s
        delay: 2s
        model: huawei
        uri: <ip>:502
        id: 2
        register:
          address: 37004
          type: holding
          decode: uint16
        scale: 0.1

DerAndereAndi, 2022-01-16:
> Huawei grid power import export (when smart power sensor is present)

Das können wir nicht erkennen? Dann müssten das wohl ein extra Template werden

RTTTC, 2022-01-16:
> 

Smart power sensor is in essence a smart power meter, that has to be orderred and installed separately, so user would know if it is present or not.

RTTTC, 2022-01-16:
I was tuning my config all night, and with this config EVCC runs fairly fast and no errors.
I am not using `Energy` meter at the moment, as I think its not needed for me, also please see here - Im not sure if thats the right register for it https://github.com/evcc-io/evcc/discussions/1928#discussioncomment-1976158
If `currents` is not utilised by evcc at the moment, probably good idea to not use it also, to speed everything up - perhaps advice users about it in template ?

Huawei grid and pv meters via Huawei Sdongle Modbus TCP connection:
```
meters:

- name: grid # Huawei grid import export power
  type: custom
  power:
    source: modbus
    timeout: 10s
    connectdelay: 1s
    model: huawei
    uri: <ip>:502
    rtu: false
    id: 1
    register:
      address: 37113 # Huawei import/export active power / IF Huawei smart power sensor is installed
      type: holding
      decode: int32
    scale: -1

#  energy:
#    source: modbus
#    delay: 1s
#    model: huawei
#    uri: <ip>:502
#    rtu: false
#    id: 1
#    register:
#      address: 37119 #  Grid active energy ???
#      type: holding
#      decode: int32
#    scale: 0.01

  currents:
    - source: modbus # Huawei phase A grid current
      timeout: 10s
      model: huawei
      uri: <ip>:502
      rtu: false
      id: 1
      register:
        address: 37107 # Huawei phase A grid current / IF Huawei smart power sensor is installed
        type: holding
        decode: int32
      scale: -0.01

    - source: modbus # Huawei phase B grid current
      timeout: 10s
      model: huawei
      uri: <ip>:502
      rtu: false
      id: 1
      register:
        address: 37109 # Huawei phase B grid current / IF Huawei smart power sensor is installed
        type: holding
        decode: int32
      scale: -0.01

    - source: modbus # Huawei phase C grid current
      timeout: 10s
      model: huawei
      uri: <ip>:502
      rtu: false
      id: 1
      register:
        address: 37111 # Huawei phase C grid current / IF Huawei smart power sensor is installed
        type: holding
        decode: int32
      scale: -0.01

- name: pv1 # Huawei Solar generation power
  type: custom
  power:
    source: modbus
    timeout: 10s
    connectdelay: 1s
    model: huawei
    uri: <ip>:502
    rtu: false
    id: 1
    register:
      address: 32080 # Active generation power
      type: holding
      decode: int32

```

DerAndereAndi, 2022-01-16:
Muss man die `ID` bearbeiten können? Was ist der Default?

RTTTC, 2022-01-16:
My experience is with dongle - I have tried to set a different Inverter id in its menu, but it does not change, always goes back to default 1
It may change in the future though, as Huawei pumps out firmware updates often, and many of the updates have some sort of improvement for modbus.
Also, some people have more than one inverter, in that case dongle would issue next id number in line (my guess)

premultiply, 2022-01-16:
Ich gehe davon aus dass das SM (als Grid) spätestens mit Batteriespeicher obligatorisch ist.

currents werden spätestens für Lastmanagement interessant, daher durchaus überall schon rein damit wo einfach möglich.

DerAndereAndi, 2022-01-16:
Also müsste die Konfig ohne Dongle dann auch ohne Batterie sein

premultiply, 2022-01-16:
Mmmh. Nicht unbedingt. Für 70%-Abregelung kann das ja auch gebraucht werden.

DerAndereAndi, 2022-01-16:
Ne, ich meine das Template ohne Dongle, das braucht dann nur "PV". Denn die Batterie kann ja ohne nicht geregelt werden

RTTTC, 2022-01-16:
Dongle is not needed to have battery in the system.
Power sensor (aka Grid meter) is necessary for the battery.

DerAndereAndi, 2022-01-16:
Yeah, but this configuration is for using Huawei inverters with the Sdongle only. At least this is what I understood

RTTTC, 2022-01-16:
I think two separate templates make sense : `grid` (with power sensor) and `pv` (without power sensor)

commit e1bbef26 (2022-01-18): Add Huawei with Sdongle

Fixes #41
closed 2022-01-18 by DerAndereAndi