evcc-io/docs #50: Better HTTP POST docs

andig, 2022-01-23:
Ah, verstanden. Das sollte gehen:

    body: { "charging_amps": ${maxcurrent} }

Die `maxcurrent` ist sowieso `int64`, deshalb ist keine zusätzliche Formatierung notwendig

_Originally posted by @andig in https://github.com/evcc-io/evcc/discussions/2309#discussioncomment-2022117_

DerAndereAndi, 2022-01-27:
Gibt es hier nur diesen Wert oder auch andere Beispiele? Wo finde ich die?

andig, 2022-01-27:
Die Werte heissen immer genau so wie das Property des custom devices (soc beim Auto etc)

andig, 2022-01-27:
Alternativ: wir ändern das so, dass der Parameter immer „value“ heisst? Wäre vielleicht intuitiver?

DerAndereAndi, 2022-01-27:
und wie wir das dann referenziert?

Wir haben hier auch 2 Doku Probleme:
1. Welche Werte gibt es
2. Wie werden die in den Plugins verwendet

Für 2. hilft das, dann ist das immer gleich.
Für 1. muss das eh in die Geräte Bereiche unter der `manual` rein, und das beispiel muss schon alle möglichen Werte beinhalten

andig, 2022-01-27:
1 hatten wir doch neulich erst diskutiert- siw sind die dokumentiert