evcc-io/docs #59: Document mincurrent/maxcurrent api

andig, 2022-01-29:
...bei Gelegenheit sollten wir mal überlegen, das in "settings" zu ändern.

andig, 2022-01-29:
`targetcharge` fehlt ebenfalls

DerAndereAndi, 2022-01-30:
Nein da bin ich anderer Meinung. Das ist eine Referenz, keine Settingsübersicht. Das ist viel zu komplex. Die möglichen Einstellungen werden in Zukunft durch die Templates alle direkt sichtbar beim Template. Das ist sowieso notwendig im irgendwann mal eine UI dafür anbieten zu können.

Das einzige was noch offen ist, sind die Bereich die heute noch nicht über Templates abgedeckt sind, z.b. generell was in Loadpoints, Messaging, etc. pp passiert. Aber da werden wir auch noch hinkommen.

Was in der Referenz klarer machen können, außer fehlendes zu ergänzen, sind die Beispiele ausführliche zu beschreiben, damit man die sich nicht aus der Hierarchie zusammendenken muss.

andig, 2022-01-30:
Verstehe ich nicht. Ist das ein Grund, bestehende APIs nicht zu dokumentieren?

DerAndereAndi, 2022-01-30:
Hatte ich missverstanden, dachte es ging um Konfigurationen. Sorry.

DerAndereAndi, 2022-01-30:
Dann verstehe ich jetzt aber deinen Text in https://github.com/evcc-io/docs/issues/59#issue-1118157650 nicht

andig, 2022-01-30:
> ..bei Gelegenheit sollten wir mal überlegen, das in "settings" zu ändern.

Ich meinte dass wir ggf. statt 10 apis mal 1 draus machen. Wir könnten uns ein "Dev meeting" Projekt anlegen um solche Themen für die Agenda zu sammeln?

commit add8638b (2022-02-05): Add minCurrent and maxCurrent to API docs

Fix #59
closed 2022-02-05 by DerAndereAndi