evcc-io/docs #76: Tariff option for selfconsumed earnings

orthoenergy, 2022-02-26:
There are some PVs starting before 2013 where selfconsumed earnings are payed  How about an option like „selfconsumed“ for them?

orthoenergy, 2022-02-27:
> I've solved this simply by specifying the price delta between self-consumption and feed-in as pv price.

Danke, das ist eine gute Idee. Leider stürzt Terminal bei Eingabe von evcc config immer ab. Mal schauen ob ich es manuell hinbekomme. 

Grüße 

Andreas Kostka

andig, 2022-02-27:
> Leider stürzt Terminal bei Eingabe von evcc config immer ab. Mal schauen ob ich es manuell hinbekomme.

Was meinst Du damit? Fehlermeldung?

orthoenergy, 2022-02-27:
Das Terminalfenster auf meinem iPad stürzt ab. Verwende Secure Term

Grüße 

Andreas Kostka

orthoenergy, 2022-02-27:
Secure Term (ios) stürzt bei ecvv configure immer ab und Terminus läuft durch aber evcc startet nicht. Ich möchte erstmal zum Testen nur meine Sonnenbatterie mit den Sensoren grid, pv, battery entsprechend der Doku einbringen, habe oben in der evcc.y’all 0.0.0. durch die IP des Raspi ersetzt aber es kommt bei Neuaufruf von EVCC configure eine Fehlermeldung zu Zeile 8.

  
GNU nano 3.2                                                                      /etc/evcc.yaml                                                                                
„
uri: 192.168.87.16:7070 # uri for ui
interval: 10s # control cycle interval

log: info

meters:
 - name: grid
     type: custom
     power:
      source: http
      uri: http://192.168.87.54:8080/api/v1/status
      jq: .GridFeedIn_W
      scale: -1 # reverse direction
 - name: pv
     type: custom
     power:
      source: http
      uri: http://192.168.87.54:8080/api/v1/status
      jq: .Production_W
 - name: battery
     type: custom
     power:
      source: http
      uri: http://192.168.87.54:8080/api/v1/status
      jq: .Pac_total_W
     soc:
      source: http
      uri: http://192.168.87.54:8080/api/v1/status
      jq: .USOC
„

Grüße

Andreas Kostka

Strasse 67 37a
13125 Berlin
Fon 030 449 29 37
Mobil 0176 4920 4930

andig, 2022-02-27:
> aber es kommt bei Neuaufruf von EVCC configure eine Fehlermeldung zu Zeile 8.

Was für eine Fehlermeldung bei konkret welchem Aufruf?

orthoenergy, 2022-02-27:
Grüße

Andreas Kostka

Strasse 67 37a
13125 Berlin
Fon 030 449 29 37
Mobil 0176 4920 4930

VolkerK62, 2022-05-26:
@andig   ist durch #128  erledigt

closed 2022-05-26 by andig