evcc-io/docs #89: Configuration example for the Polestar 2

ThomDietrich, 2022-03-16:
Hey there!

I want to add the Polestar 2 to the documentation of supported cars. Does anyone already have a working tested config or can share relevant experience?

To my knowledge it's currently not possible to access the car directly, so the manual route has to be used. I found a longer discussion including a supposedly working example for OpenWB: https://polestar.fans/t/polestar-2-openwb-soc-modul/5857/7

I can start testing the Polestar 2 charging coordinated by evcc in the next couple of weeks and will add my learnings here.

The result of this discussion should be a pull request to document the Polestar 2.

andig, 2022-03-16:
We don‘t have a manual module, hence Polestar remains unsupported until someone can provide an Api.

ThomDietrich, 2022-03-16:
Hey @andig 
I'm slightly confused now. What about https://docs.evcc.io/docs/devices/vehicles#manuell ?
As long as there is no API, what is the best way to work with evcc and the Polestar then?

ThomDietrich, 2022-03-16:
Right, I can see that now. So even if I went with the "javascript" plugin type I would need it to produce developing SoC/Status/Range information - which of course I can't. I didn't look at the details behind the openWB implementation but I thought it works with a combination of coulomb counting, manual inputs and/or some internal sensing. evcc does not do anything like it.

Sorry for the confusion. With that cleared up, what's your suggestion of how to configure `vehicles:`?

Going forward, what do you think would need to be done to add better evcc support for any car model without an api?

Thanks

andig, 2022-03-18:
> Going forward, what do you think would need to be done to add better evcc support for any car model without an api?

M.E. gar nichts- evcc funktioniert ja auch ohne ;)

Für weitere Fragen bitte Discussions verwenden- das docs Repo ist der falsch Platz dafür.

closed 2022-03-18 by andig