evcc-io/docs #93: Document maxGridSupplyWhileBatteryCharging

andig, 2022-03-27:
https://github.com/evcc-io/evcc/pull/3015

VolkerK62, 2022-06-04:
@andig  Soweit ich das ermittelt haben, ist das ein optionaler parameter in der site (so wie z.B.  `residualpower`) und der kommt zum Einsatz bei Hybrid-Wechselrichtern. Ist das soweit schon mal richtig? Um das genaue Problem zu beschreiben, werde ich mich noch ein wenig einlesen müssen.

VolkerK62, 2022-06-04:
`maxGridSupplyWhileBatteryCharging`
Dieser Parameter kann bei Hybrid-Wechselrichtern zum Einsatz kommen. Sollte die DC-Leisung größer sein, als die AC-Leistung des Wechselrichter, kann das während der Batterieladung dazu führen, dass zur Fahrzeugladung zusätzlich Netzstrom bezogen wird. Mit diesem Parameter kann in dem Fall der maximale Netzbezug begrenzt werden.

Wäre das eine korrekte Beschreibung? 
Welche Werte sind für den Parameter zulässig? 
positiv=Netzbezug, negativ= Einspeisung?

andig, 2022-06-04:
Dabei gehts um https://github.com/evcc-io/evcc/issues/2734. Das zu lösende Problem ist, dass die Batterieladung teilweise aus Leistung besteht, die NICHT anderweitig ins Netz eingespeist werden könnte (mangels AC Leistung des WR). Um an diese Grenze ran zu kommen versucht man jetzt nicht die Batterieladung auf Null zu fahren, sondern den Netzzähler bis zu minimalem Bezug. In dem Moment ist sichergestellt, dass die komplette AC Leistung der Batterie auf Null ist.

andig, 2022-06-04:
https://github.com/evcc-io/evcc/pull/2735/files#diff-71cfdec6af7c316e7ecebdaaaa823cdd573fc6707e493ff7b01376903e6d0dadR29

VolkerK62, 2022-06-04:
Was ich noch verstehe: 
EVCC schaut doch eigentlich auf grid. Warum führt der o.g. Umstand dann zu Netzbezug? 
Ist in dem Fall auch der grid Wert falsch, weil alles vom gleichen System gemessen wird?

Sorry, dass ich möglicherweise blöde Fragen stelle.
Evtl. ist das Thema zu hoch für mich.

premultiply, 2022-06-04:
Das ist eine ganz spezielle Situation die nur bei Hybrid-WR auftreten kann bei denen die interne DC-Leistung PV->BAT+AC höher ist als die Leistung des AC-Wechselrichterteils zum Netz hin.

Normalerweise wird die Einspeisung+Batterieladeleistung(+Autoladeleistung) von evcc ja als "verfügbare" Autoladeleistung gezählt. Also speziell bei der Batterie Leistung die man dieser wegnehmen kann um im Gegenzug die Leistung der Autoladung zu erhöhen.

In diesem Spezialfall funktioniert das aber nicht, da der AC-Pfad des Hybrid-WR schon mit der PV-Erzeugung leistungsmäßig am Anschlag ist, so dass der verbleibende Teil "direkt" auf dem internen DC-Pfad in der Batterie landet.
Obwohl evcc hier eine "Batterieeinspeisung" sieht kommt es aber nicht dort heran, da der AC-Pfad bereits ausgelastet ist.

Beispiel 10 kW Hybrid-WR:
PV-Erzeugung: 15 kW
Netzeinspeisung: 10 kW
Batterieladung: 5 kW

In diesem Fall sieht evcc 10 kW Netzeinspeisung + 5 kW Batterieladung = 15 kW mögliche Ladeleistung.
Wird die Ladung nun mit 15 kW begonnen stellt sich am Netzanschluss aber nicht 0 ein sondern ein Netzbezug von 5 kW da der WR eben schon mit 10 kW am Anschlag ist.

Ich hoffe das war einigermaßen verständlich?

VolkerK62, 2022-06-05:
Soweit habe ich das verstanden. Wo es noch klemmt, ist:

> Normalerweise wird die Einspeisung+Batterieladeleistung(+Autoladeleistung) von evcc ja als "verfügbare" Autoladeleistung gezählt. 

Das passiert doch aber nur, wenn `prioritySoc` erreicht ist. Oder ist das in dem speziellen Fall anders?

Durch diesen Parameter `mGSWBC` wird also ein Grid-Schwellenwert gesetzt, der, wenn er überschritten ist, die Batterieladeleistung für die "site power" Berechnung auf null setzt und dadurch weniger Autoladeleistung zur Verfügung steht. (Also praktisch so, als wäre `prioritySoc` noch nicht erreicht.)

Hier sind nur positive Werte möglich/sinnvoll?
Habt ihr eine Empfehlung?  Ich habe sehr oft 50W gelesen. Das wären dann 50W Netzbezug?
Inwieweit ist hierbei ein Wert bei `residualpower` zu berücksichtigen?

Jetzt mal abseits der Doku:
Wird dadurch nicht eine PingPong Regelung erschaffen?
Schwellwert ist überschritten -> Batterieladung geht berechnungsmäßig auf null -> Wallbox regelt runter -> Schwellwert wird erreicht -> Batterieladung zählt wieder mit - > Wallbox regelt hoch -> dann geht es wieder von vorne los.

andig, 2022-06-05:
> Das passiert doch aber nur, wenn prioritySoc erreicht ist.

Genau. Bis dahin hat die Batterie ja sowieso immer Vorrang, wird aus der Rechnung also rausgenommen.

> Hier sind nur positive Werte möglich/sinnvoll?

Ja

> Durch diesen Parameter mGSWBC wird also ein Grid-Schwellenwert gesetzt, der, wenn er überschritten ist, die Batterieladeleistung für die "site power" Berechnung auf null setzt und dadurch weniger Autoladeleistung zur Verfügung steht. 

Das Beispiel von Premultiply erklärt es besser. Siehe auch Kommentar in https://github.com/evcc-io/evcc/blob/master/core/helper.go#L33.

Wenn wir Netzbezug sehen, *obwohl* die Batterie geladen wird, dann wissen wir dass die Batterie rein DC-DC geladen wird weil der AC Pfad des WE ausgelastet ist. Genau dann steht auch diese Batterieladeleistung nicht zur Verfügung und wird daher für die Berechnung genullt. Da auch die WR nicht auf genau 0W und in Millisekunden regeln gibts dafür einen Schwellwert.

VolkerK62, 2022-06-05:
> **Update:** aber die Fragen sind gut. Ich musste auch erstmal wieder den Knoten aus dem Kopf bekommen.

Schön, dass ich dir beim Gehirnjogging behilflich sein kann ;-)

Ich werde mir mal Gedanken machen, wie man das EINFACH erklären kann.

andig, 2022-06-05:
> Ich werde mir mal Gedanken machen, wie man das EINFACH erklären kann.

Der war gut :). Vielleicht helfen ja die Diagramme aus dem ursprünglichen Issue dabei.

VolkerK62, 2022-06-05:
Wäre das technisch korrekt und verständlich?

Dieser Parameter kommt bei Hybrid-Wechselrichter Systemen zum Einsatz, bei denen die DC-Erzeugungsleistung größer als die AC-Ausgabeleistung des Wechselrichters ist.
Beispiel: 15kW PV-Erzeugungsleistung, aber nur 10kW AC-Ausgabeleistung des WR = 5kW Batterieladeleistung auf DC-Ebene.
Wenn prioritySoc erreicht ist, wird die gesamte Batterieladeleistung als verfügbare Fahrzeugladeleistung betrachtet. In dem obigen Beispiel käme es dann zu Netzbezug, weil EVCC die 5kW zur verfügbaren Fahrzeugladeleistung addieren würde, obwohl diese vom Wechselrichter dem Netz gar nicht zur Verfügung gestellt werden können. 
Mit diesem Parameter kann ein Schwellenwert für den Netzbezug gesetzt werden, der bewirkt, dass diese Batterieladeleistung nicht als verfügbare Fahrzeugladeleistung angesehen wird.
Empfohlen wird ein Wert von mindestens 50. Je nach Trägheit der betroffenen Regelungssysteme kann er auch höher sein.

andig, 2022-06-05:
Der eigentliche Sinn erschließt sich noch nicht so richtig. Auch den Bezug zu prioritySoC finde ich schwierig- das ist ja orthogonal. Was Besseres fällt mir aber auch nicht ein...

VolkerK62, 2022-06-05:
> Mit diesem Parameter kann ein Schwellenwert für den Netzbezug gesetzt werden, der bewirkt, dass diese Batterieladeleistung nicht als verfügbare Fahrzeugladeleistung angesehen wird.

Das ist m. E. der Sinn des Parameter. Der Rest ist nur der Versuch einer Erklärung, warum der Parameter notwndig ist.
Priority Soc hab ich mit reingenommen, da unterhalb der Effekt ja gar nicht auftritt.

Vorschlag: Ich bau den PR. Wenn dann Nachfragen zur Doku kommen, sieht man ja, was u. U. noch anders/besser erklärt werden muss.

PR #154 (2022-06-05): Update site.md
fix #93 
schwere Geburt :-)
commit b7c74d1f (2022-06-05): Update site.md (#154)

fix #93 
schwere Geburt :-)
closed 2022-06-05 by premultiply