evcc-io/docs #102: API Doku falsch - loadpoint index

Waldmensch1, 2022-04-14:
hier https://docs.evcc.io/docs/reference/api#mqtt-api

steht
`Die MQTT API folgt der REST API Struktur, mit den Ladepunkt (loadpoint) IDs bei 0 beginnend:`

die loadpoint ID's beginnen allerdings im MQTT publisher bei 1

vgl mqtt.go zeile 137ff
```
		if p.LoadPoint != nil {
			id := *p.LoadPoint + 1
			topic = fmt.Sprintf("%s/loadpoints/%d", m.root, id)
		}
```

premultiply, 2022-04-14:
Machst du einen PR für die notwenigen Doku-Änderungen hier? Danke!

andig, 2022-04-14:
Ich denke auch- wir sollten hier lieber die Doku ändern als die Implementierung?

Waldmensch1, 2022-04-14:
Ich habe mir das Doku repo geforked und würde einen PR machen, auch für die identifier Liste. Betrifft die id 0/1 Sache nur MQTT oder die REST Api auch?

naltatis, 2022-04-16:
> Betrifft die id 0/1 Sache nur MQTT oder die REST Api auch?

Bei REST beginnt es mit `0`.

closed 2022-04-16 by naltatis