evcc-io/docs #117: WARP-charger /evse/state outdated

ben-bole, 2022-05-20:
Hey,
I am currently having problems with the connection between my WARP-charger (hardware-version 1) an evcc (0.92).
evcc gives the following error message:

> charger: warp/wb/evse/state outdated: 12m4s

I believe that's due to the fact, that the WARP callbox only pushes values on /evse/state when changed, referring to a community post by one of the Tinkerforge devs. (`https://www.tinkerunity.org/topic/6317-mqtt-intervall/?do=findComment&comment=40036`) I can verify this behaviour by monitoring the MQTT-topic. The topic that indeed is updated every second ist /evse/low_level_state
In the Warp-admin intiferace I have set the MQTT interval on the default value of 1 second.

Does anyone else experienced this issue or might the problem root somewhere else.

Thanks in advance, Benedict

My charger-part of the evcc.yaml:

```
chargers:
  - name: WARP
    type: template
    template: tinkerforge-warp
    fw2: # Firmware v2 installiert
    host: 10.32.8.9 # IP Adresse oder der Hostname des MQTT Brokers
    port: 1883 # MQTT Broker Port # Optional
    topic: warp/wb  #Topic (ohne / am Anfang) # Optional
    timeout: 30s # Akzeptiere keine Daten die älter als dieser Wert sind # Optional
```

andig, 2022-05-20:
Welche FW hast Du drauf?

ben-bole, 2022-05-20:
Die allerneueste: 2.0.5-627e4d9d

andig, 2022-05-20:
Na dann… solltest Du das in der Config vllt auch eintragen ;) 

    fw2: true

closed 2022-05-20 by andig

ben-bole, 2022-05-20:
Top! Danke! Ich hab' dafür in der Doku gleich mal nen PR gemacht :)

andig, 2022-05-20:
Voll nett, aber falsche Stelle. Die Doku ist aus den Templates generiert. Da muss die Änderung im normalen evcc Repo rein.