evcc-io/docs #118: Remove types?

andig, 2022-05-20:
@naltatis was machen wir mit diesen Listen: https://docs.evcc.io/docs/reference/configuration/chargers

Die sind nicht aktuell und ohne Parameter eigentlich auch nicht hilfreich. Entfernen???

andig, 2022-05-22:
Siehe auch https://github.com/evcc-io/evcc/issues/3455, das führt zu Verwirrung.

naltatis, 2022-05-22:
Ja, ich würde vorschlagen, dass wir als öffentliche Schnittstelle nur noch `custom` und `template` hier in der Doku haben. Die direkte Nutzung der andern Typen würde ich inzwischen als umdokumentierte Private API sehen.

VolkerK62, 2022-05-31:
Kurze Nachfrage: hier  https://docs.evcc.io/docs/devices/chargers  sind alle aufgeführt, die implementiert sind und für alle gibt es auch ein entsprechendes template?  Oder sollte ich das vorher kontrollieren?  Der entsprechende Link ist ja hier https://docs.evcc.io/docs/reference/configuration/chargers  schon enthalten. ich würde dann die Liste kürzen bis auf `custom`.  Bei einigen gibt es Hinweise z.B.  "CFos braucht ein meter", derartige Hinweise würde ich dann in die Geräteliste übernehmen. Oder ist das nicht nötig, da das Template entsprechend gebaut ist?

VolkerK62, 2022-05-31:
und was mir gerade noch aufgefallen ist, eigentlich sollte das gleiche Prinzip mit dem Templates doch auch bei  https://docs.evcc.io/docs/reference/configuration/meters   und https://docs.evcc.io/docs/reference/configuration/vehicles  angewendet werden, oder?

andig, 2022-06-03:
> Kurze Nachfrage: hier https://docs.evcc.io/docs/devices/chargers sind alle aufgeführt, die implementiert sind und für alle gibt es auch ein entsprechendes template? Oder sollte ich das vorher [...]

Die Templates dienen eigentlich "nur" dazu, die internen Typen mit Zusatzinformationen für die Konfiguration anzureichern, als auch komplexere Plugins "zusammenzustöpseln". Bisher ist noch nicht jedes Gerät in ein Template verpackt. Bei manchen (`movingaverage` meter) ist das auch nicht sinnvoll möglich.

Alle die schon mit einem Template hinterlegt sind können aber von der types Seite verschwinden!

VolkerK62, 2022-06-03:
Ich habe keinen Type gefunden, der nicht in der Geräteliste aufgeführt ist. Ich habe den Hinweis auf die Geräteliste erweitert und nur noch `custom` drin gelassen.

PR #144 (2022-06-03): Update chargers.mdx
fix #118
commit d8ce7e71 (2022-06-03): Update chargers.mdx (#144)

fix #118