evcc-io/docs #131: Standardwerte für enable / disable bei loadpoints

DanielGlaas, 2022-05-28:
Die Einstellungen für Aktivierung und Deaktivierung der Ladung via PV können über die Parameter [enable](https://docs.evcc.io/docs/reference/configuration/loadpoints/#enable) und [disable](https://docs.evcc.io/docs/reference/configuration/loadpoints/#disable) durchgeführt werden.

Bei beiden Werte gibt es jeweils `threshold` und `delay` als Unterwerte. Für diese sind keine Werte als Standard angegeben, was allerdings hilfreich wäre.

premultiply, 2022-05-28:
Wichtig ist auch dass die entsprechenden Zeiteinheiten immer mit angeben werden! ;)

andig, 2022-05-28:
> Für diese sind keine Werte als Standard angegeben, was allerdings hilfreich wäre.

Was meinst Du damit? Ich sehe hier jede Menge Defaults: https://github.com/evcc-io/evcc/blob/master/evcc.dist.yaml#L107

DanielGlaas, 2022-05-28:
Also das ist die Ansicht für den Wert `enable.threshold`:
[image]
Unter `delay` für `enable` steht auch kein Standardwert
[image]

Bei `disable` steht ein nicht passender Wert darunter. 
[image]

> Ich sehe hier jede Menge Defaults: https://github.com/evcc-io/evcc/blob/master/evcc.dist.yaml#L107

Diese Werte sehe ich in der Doku nicht.

DanielGlaas, 2022-05-28:
Wird diese Seite mit den Beschreibungen und Standardwerten etc. automatisch generiert oder manuell?
Mit einem entsprechendem Hinweis könnte ich versuchen, mich an dieser Stelle einzubringen.

premultiply, 2022-05-28:
Sehr gerne. Diese Seite ist rein manuell. Nur die Seite(n) mit den Geräten werden automatisch aus den Templates erzeugt.

DanielGlaas, 2022-05-28:
Ok, dann kann ich das ja mal versuchen, evtl. morgen Abend.

Ich clone mir diese Datei: https://github.com/evcc-io/docs/blob/main/docs/reference/configuration/loadpoints.md
und mache dann einen Pull-Request mit den Änderung. Ist das das korrekte Vorgehen?

premultiply, 2022-05-28:
Auf gut Deutsch: Du kannst dir das ganze docs-Repository clonen, dann darin lustig rumeditieren und dann anschließend aus den ganzen commits einen PR formen. Das geht mit den entsprechenden git-Tools via Konsole, Anwendung oder z. B. auch in VSCode.

Oft wesentlich einfachere Alternative: Einfach in github (Webseite) die jeweilige Seite bzw. Datei bearbeiten (Symbol oben rechts) dann wird am Ende automatisch ein PR draus. Für die kleine Sachen und die Doku reicht das allemal. Da sind wir auch nicht so formalistisch wie im evcc-Quelltext.

DanielGlaas, 2022-05-28:
Ok, alles klar. Ich nehme mir vor, es mal auszuprobieren.

PR #137 (2022-06-03): Update loadpoints.md
This PR adds default values to the explanations of the settings of the loadpoints configuration.

The inserted values are taken from https://github.com/evcc-io/evcc/blob/master/evcc.dist.yaml

This fixes #131.
commit 06f6f8cb (2022-06-03): Update loadpoints.md (#137)

This PR adds default values to the explanations of the settings of the loadpoints configuration.

This fixes #131.
closed 2022-06-03 by premultiply