evcc-io/docs #161: FAQ: connection refused

VolkerK62, 2022-06-14:
kommt aus https://github.com/evcc-io/evcc/discussions/3620

willst du vllt mal i/o timeout und connection refused in die Faq aufzunehmen? Die Ursache ist ja immer gleich. Ob ein Gerät auf ping antwortet oder das UI aufrufbar ist ist dabei völlig irrelevant. Telnet host port kann zur Diagnose verwendet werden.

VolkerK62, 2022-06-14:
@andig  Wie ist das? Ihr habt in den Templates entsprechende Standardports hinterlegt und wenn am Gerät ein anderer Port eingestellt ist, kommt "i/o timeout" und "connection refused"?
Gibt es eine Liste der Ports oder sind die von Gerät zu Gerät unterschiedlich?

premultiply, 2022-06-14:
Mal ein paar Hintergründe für dich:

Die meisten Protokolle benutzen von sich aus übliche Standardports. HTTP z. B. 80 oder auch Modbus 502.
D.h. aber nicht unbedingt dass diese auch immer genutzt sind. evcc selbst öffnet z. B. per Default für seine eigene Web-UI für HTTP den Port 7070.
Wo immer Hersteller eigene "Standardports" für bestimmte Protokolle verwenden haben wir diese in die Templates eingebaut. Dort kann man sie entweder übernehmen oder manuell überschreiben falls die Gerätekonfiguration vom üblichen Auslieferungszustand abweicht.
Bei manchen WR wird z. B. für Modbus ab Werk Port 1502 statt 502 verwendet. Warum auch immer.

Die Meldungen "i/o timeout" oder "connection refused" besagen aber eigentlich etwas anderes:
Bei "connection refused" versucht evcc eine Verbindung zu einem Gerät mit der konfigurierten Adresse sowie dem angegeben Port herzustellen, das Gerät verweigert aber aktiv den Verbindungsaufbau. Dies ist z. B. regelmäßig bei Solaredge-Geräten der Fall wo wohl standardmäßig der interne Modbus-Server deaktiviert ist. Daher kann keine Verbindung aufgebaut werden bis der Modbus-Server in der WR-Konfiguration aktiviert wurde. Lästiger- und unnötigerweise kann das dort wohl auch nur der Hersteller auf Anfrage tun - aber das ist eine andere Geschichte. Davon abgesehen lässt der Modbus-Server dort auch nur eine einzige parallele Verbindung zu. Weitere Verbindungswünsche werden ebenfalls abgelehnt falls schon eine Verbindung (z. B. von einem anderen System oder einer anderen Instanz) besteht.
Generell kann man sagen: Die Gegenstelle (also das Gerät) ist zwar prinzipiell wohl erreichbar lehnt einen Verbindungswusch auf dem Port mit einer entsprechenden allgemeinen Meldung ab.

"i/o timeout" ist etwas spezieller. Hier besteht i.d.R. schon mal eine Verbindung, ein über diese Verbindung gesendetes Kommando liefert aber innerhalb eines Timeouts keine Antwort zurück. Das kann viele Gründe haben.
Entweder das Gerät hat schon die Anfrage überhaupt nicht verstanden (oder bekommen) und kann daher nicht antworten, es ist überlastet oder ausgefallen und kann daher nicht (schnell genug) antworten oder die Verbindung selbst ist so schlecht oder langsam das Daten nur verzögert oder gar nicht übertragen werden. Wie auch immer: Die anfragende Instanz (hier evcc) bekommt in einer angemessenen Zeit (Timeout) keine Antwort auf eine Anfrage bzw. ein Kommando an eine Gegenstelle (Gerät bzw. System).

PR #162 (2022-06-14): Update setup.mdx
fix #161
closed 2022-06-14 by naltatis