evcc-io/docs #164: Menüpunkt "Anleitungen" umbenennen

VolkerK62, 2022-06-15:
Ich habe leider keine Ahnung, ob und wie ICH das ändern könnte.
Ich habe ja den früheren Unterpunkt FAQ thematisch auf die anderen Unterpunkte aufgeteilt.
Deshalb sollte der Menüpunkt jetzt m. E. umbenannt werden.
Mein Vorschlag wäre "Tipps und FAQ".

VolkerK62, 2022-06-21:
@naltatis  kannst du mir hier weiterhelfen?

naltatis, 2022-06-21:
Den Titel für den Menüpunkt kann man hier ändern:
https://github.com/evcc-io/docs/blob/main/docs/guides/_category_.json

Die Unterpunkte werden automatisch anhand der md(x) Dateien im [Verzeichnis](https://github.com/evcc-io/docs/tree/main/docs/guides) erstellt. Ich würd vorschlagen die `faq.mdx` zu einer `wallbox.mdx` und `meters.mdx` zu kopieren und die Ausgangsdatei dann zu löschen. Die Navieinträge entsprechen im Standardfall der Hauptüberschrift der Markdown-Dateien.

VolkerK62, 2022-06-21:
ich hoffe ich habe dich richtig verstanden und nix kaputt gemacht ;-)