evcc-io/docs #178: Document new apis

andig, 2022-06-26:
https://github.com/evcc-io/evcc/pull/3721

closed 2022-06-27 by premultiply

VolkerK62, 2022-06-27:
@premultiply  sorry für die unverschämte Frage ;-)   
Ist es richtig, dass bei den neuen REST API kein `site` dabei ist? (im Gegensatz zu MQTT)

andig, 2022-06-27:
Nein... es steht nur nicht explizit dabei.

premultiply, 2022-06-27:
Ich habe mich am Quellcode orientiert und da sehe ich zumindest keine Site:
https://github.com/evcc-io/evcc/blob/8bc7862fa3376a8e2e2969c5384efd23ba42cb8e/server/http.go#L44-L52

andig, 2022-06-27:
Die markierten sind alle site, loadpoints kommen etwas weiter unten. Vielen Dank @premultiply !

VolkerK62, 2022-06-27:
Wie muss man das aufrufen?   `api/residualpower` oder `api/site/residualpower`

andig, 2022-06-27:
Genau so wies jetzt dokumentiert ist

commit 96244067 (2022-06-27): Update api.md

Fixes #178