evcc-io/docs #181: Nachziehen der Lademodi

DanielGlaas, 2022-07-18:
In evcc in Version 0.96 gibt es die Modi

- Aus
- Min + PV
- PV
- Schnell
Siehe dieser Screenshot:
[image]

In der Doku steht aber "Schnell" nicht beschrieben, sondern "Sofort"
[image]

Leider habe ich in den Release Notes von evcc keinen Hinweis gefunden, wann und warum das geändert wurde:
https://github.com/evcc-io/evcc/releases
Ansonsten hätte ich mich selbst an einer Anpassung versucht.

andig, 2022-07-18:
@naltatis in welcher Richtung wollen wir das fixen?

closed 2022-07-18 by premultiply

naltatis, 2022-07-19:
Danke für die Anpassung!

Ja, ist im Zuge der Darkmode Umstellung reingekommen. Die Idee ist hier auf die Ladegeschwindigkeit (schnell) zu gehen und nicht auf das Ladeverhalten. Auch die Reihenfolge hat sich geändert (vorher Aus, Sofort, Min+PV, PV) nun von langsam nach schnell (Aus, Min+PV, PV, Sofort). Min+PV wird auf kurz oder lang aus der Moduswahl rausgenommen werden (siehe https://github.com/evcc-io/evcc/discussions/3530) und das Verhalten anderweitig auswählbar gemacht. Bis dahin würde ich aber PV und Min+PV auch noch mal tauschen, damit das mit der aufsteigenden Geschwindigkeit / Erwartete Energiemenge passt.

PR #182 (2022-07-18): Update charging.mdx
fix #181 
Modus `sofort` in `schnell` geändert