evcc-io/docs #189: FAQ Struktur

andig, 2022-08-03:
Ich hab mit schwer getan, für https://docs.evcc.io/docs/guides/faq#bei-meinem-peugeotopel-wird-der-ladezustand-nur-aktualisiert-wenn-ich-die-hersteller-app-benutze den richtigen Platz zu finden. Einerseits ist es ein Fahrzeugthema. Andererseits finde ich unter dem Fahrzeugartikel eher die grundsätzliche Nutzung von evcc erklärt (was gut ist). Dann aber wiederum mit einzelnen Fragen gemischt (stilistisches Mittel)?

Können wir das sinnvoll trennen:
- Benutzerhandbuch (matter-of-fact Formulierungen)
- FAQ (prominent an zentraler Stelle)

Was meint ihr?

/cc @VolkerK62 @premultiply

VolkerK62, 2022-08-03:
Mein erster Gedanke war, das unter "FAQ: Sonstiges" aufzunehmen. Da steht es jetzt auch schon drin. 
Den Menüpunkt könnte man doch als Sammler für so spezifische Themen nehmen.
Wenn es mehr werden, könnte man auch unter Fahrzeuge, solche Hersteller-spezifischen Umstände als Unterpunkte aufnehmen.

VolkerK62, 2022-08-03:
Ich habe mir das nochmal durch den Kopf gehen lassen. Ich finde, wir sollten das unter FAQ/Fahrzeuge aufnehmen. Überschrift: "Herstellerspezifische Besonderheiten" und dann entsprechende Unterpunkte mit den Herstellern.
Dann könnte man auch z.B. auf das 1p/8A Problem der ZOE hinweisen.
Analog dazu gibt es bestimmt auch Besonderheiten bei den Wallboxen und der Hausinstallation? 

Ein "Benutzerhandbuch" existiert eigentlich schon unter "Referenz" mit den ganzen Einstellungsmöglichkeiten.

closed 2024-03-20 by naltatis