evcc-io/docs #190: Document vehicle

andig, 2022-08-04:
> Wenn `vehicle` wirklich so wesentlich anders funktioniert als `vehicles`, würde ich empfehlen, den Parameter umzubenennen in vehicleHardOverride o.ä., und in der Dokumentation vehicles als Standardfall zu zeigen.

/cc @VolkerK62 @premultiply  die Bedeutung von `vehicle` hat sich ja tatsächlich in `defaultVehicle` geändert. Doku Update oder Entwicklung?

/cc @mlnoga

VolkerK62, 2022-08-04:
Ist in der Doku richtig beschrieben. vehicle = DefaultVehicle. 
Die Verwechselung (ob mit oder ohne "s" am Ende) ist natürlich groß. 
Insofern wäre ich dafür, den Namen des Parameter von `vehicle` in `defaultVehicle` zu ändern. Dann wäre es klarer.

VolkerK62, 2022-08-14:
Durch das Redesign Fahrzeugerkennung gibt es ja jetzt nur noch `vehicle`. Issue zum Doku Update mit den zu ändernden Punkte ist erstellt.

closed 2022-11-17 by andig