evcc-io/docs #192: FAQ: add minsoc

andig, 2022-08-07:
> Ich frage weil ist mehrmals vorgekommen das der SOC nicht erkannt wurde und dann mit voller 11kw geladen.

Das Problem kommt immer wieder hoch

VolkerK62, 2022-08-07:
Soweit ich das verstehe, ist kein SoC = SoC 0%.
Und wenn SoC unter MinSoC ist, wird volle Pulle geladen.
Lässt sich also verhindern, indem man `minsoc: 0` definiert?

Sollte das dann bei `vehicle` oder beim `loadpoint` (oder bei beiden) eingetragen werden?

andig, 2022-08-07:
Gar nicht definiert ist auch 0. Wo einzutragen hängt davon ab wann es gelten soll. Wir könnten auch mal überlegen, die min und target soc Loadpoint Settings abzuschaffen. Egtl sind das ja Eigenschaften des Fahrzeugs?

VolkerK62, 2022-08-07:
Kann es Fälle geben, in denen das Fahrzeug nicht erkannt, aber ein SoC übermittelt wird? Ansosnten machen diese EInstellungen am Loadpoint eigentlich keinen Sinn.
Ist das noch ein Relikt aus "alter Zeit"? Gab es die bevor es diese Parameter beim Fahrzeug gab?

andig, 2022-08-07:
Ja, absolut. Die sind egtl. Alle aus alter Zeit. Bleibt die Frage des Handlings für den Anwender. Das sollten wir in neuen Issue besprechen.

premultiply, 2022-08-07:
Waren die LP-Werte nicht der Basissatz der beim Gastfahrzeug angewendet wird?

andig, 2022-08-08:
Mindestens target und min soc sind für ein nicht auslesbares Gastfahrzeug aber ohnehin irrelevant- was bleibt da noch?

PR #191 (2022-08-08): Update charging.mdx
fix #185
fix #192
closed 2022-08-08 by premultiply