evcc-io/docs #225: Raspi Pi : Anleitung für Installation des downloaded evcc file

PV-Freak, 2022-09-30:


Sillium007, 2022-09-30:
Wenn du ein kompatibles System hast ist es am einfachsten über apt. 
Ansonsten wie in der Anleitung beschrieben:
https://docs.evcc.io/docs/installation/manual/

Beim anderen Thread ging es ja um ein Downgrade.. Aber ja.. Upgrade würde auch entsprechend gehen. Halt am besten vorher ne Sicherung machen falls es nicht klappt.
Und die Änderungen beachten. Hab welche die eine Änderung an der Config erforderten.

PV-Freak, 2022-10-01:
Hallo Sillium007
Vielen Dank für deine ausführliche Antwort.
Nun habe ich endlich gechecked, dass die „manuelle“ Installation unter deinem Link ja gut beschrieben ist. 
Auf die Schnelle habe ich in der evcc Anleitung auf der Suche nach der manuellen Installation immer nur WINDOWS gesehen und nicht den ganzen Text  „Manuell (inkl. Windows)“ gelesen 😅….
Beste Grüsse
PV-Freak

PR #243 (2022-11-17): Korrektur "Überschrift" zum besseren Finden
fix #225
commit 094ab199 (2022-11-17): Korrektur "Überschrift" zum besseren Finden (#243)

fix #225
closed 2022-11-17 by premultiply