evcc-io/docs #226: sign()-Funktion dokumentieren

premultiply, 2022-09-30:
Ref https://github.com/evcc-io/evcc/pull/4494

VolkerK62, 2022-11-17:
nachdem ich die mathematische Signum-Funktion jetzt (hoffentlich) verstanden habe ...
(Jede positive Zahl wird zu +1, jede negative zu -1 und 0 bleibt 0)

das ist doch für das calc-plugin, oder?
für welche use cases bräuchte man das?

andig, 2022-11-17:
Wenn Du einen vorzeichenbehafteten Wert hast (Phasenleistung) aus dem Du einem Absolutwert ein Vorzeichen verpassen willst (Phasenstrom) machst Du eine Multiplikation mit dem Vorzeichen des anderen Wertes.

VolkerK62, 2022-11-17:
Ah, um z.B. aus dem Shelly 3EM die korrekten Phasenströme zu bekommen (Lastmanagement).

Und wie sähe die Syntax aus?

```
source: calc
mul:
- sign:
    source: http
      uri: http://192.168.178.27/cm?cmnd=Status%208
      jq: .StatusSNS.SML.Power
- source: http
    uri: http://192.168.178.27/cm?cmnd=Status%208
    jq: .StatusSNS.SML.Current

```

andig, 2022-11-17:
Mul braucht mehrere Sources- die erste ist dann halt wieder ein calc.

VolkerK62, 2022-11-17:
ich habe mal hier beim SDM630 gespickelt ;-)  https://github.com/evcc-io/evcc/pull/5131

Dann schau ich mal, wie ich das in der Doku unterbringe.

```
source: calc
mul:
- source: calc
  sign:
    source: http
      uri: http://192.168.178.27/cm?cmnd=Status%208
      jq: .StatusSNS.SML.Power
- source: http
    uri: http://192.168.178.27/cm?cmnd=Status%208
    jq: .StatusSNS.SML.Current
```

andig, 2022-11-17:
Top 👍🏻

PR #244 (2022-11-18): add sign zu calc plugin
fix #226
commit cf63eb36 (2022-11-18): add sign zu calc plugin (#244)

fix #226
closed 2022-11-18 by premultiply