evcc-io/docs #228: Doku CoarseCurrent

VolkerK62, 2022-10-01:
>Lösung:
add
`features: ["coarsecurrent"]`
to the vehicle config.
/cc @VolkerK62

und was trägt man da ein? true/false?

_Originally posted by @VolkerK62 in https://github.com/evcc-io/evcc/issues/4662#issuecomment-1264203297_

andig, 2022-10-01:
Features ist eine Liste von Strings, ich hab einfach die einzeilige Notation verwendet

VolkerK62, 2022-10-01:
Bahnhof, Koffer weg .... ;-)   Du hast vergessen, dass ich kein Hacker bin ...

ich habe verstanden, dass das für die Kombination mA(mpere)-Wallbox und A(mpere)-Auto sinnvoll ist.

Muss man da als Anwender tätig werden?
Wenn ja, trägt man das in der Konfig beim vehicle ein?
Wenn ja, in welcher Form?

andig, 2022-10-01:
Ok ;). Also ja/ja/genau wie es oben steht (deshalb hab ichs ja hingeschrieben!). Geht aber NICHT mit den Templates sondern nur wenn das Auto „von Hand“ konfiguriert ist (type: tesla). Wer an die Stelle kommt und verstanden hat, dass seine Regelung zappelt, der ist dann auch in der Lage das zu konfigurieren…

PR #232 (2022-10-05): Doku: `coarsecurrent` ergänzt
fix #228
commit 03e6a69e (2022-10-05): Doku: `coarsecurrent` ergänzt (#232)

* Doku: `coarsecurrent` ergänzt

fix #228

* Update vehicles.mdx
closed 2022-10-05 by premultiply