evcc-io/docs #241: docker-compose needs volume /etc/machine-id mounted

fellek, 2022-11-03:
since evcc 0.106.0 in docsker-compose mounting of /etc/machine-id seems to be mandatory. Otherwise evcc is crashing with error:
`could not get machineid: open /etc/machine-id: no such file or directory; for manual configuration use plant: [container-id]`
      -  please add this line into standard-configuration. 
`      - /etc/machine-id:/etc/machine-id`
Currently its only marked as SMA / EEBus Special configuration.
[Docker Installation > Docker-Compose](https://github.com/evcc-io/docs/blob/main/docs/installation/docker.mdx)

premultiply, 2022-11-03:
Docs PR welcome

andig, 2022-11-03:
No PR. It's a bug https://github.com/evcc-io/evcc/pull/5029 and the docker release has been unpublished.

closed 2022-11-03 by andig