evcc-io/docs #245: log-level in Doku vervollständigen

VolkerK62, 2022-11-21:
@andig welche evcc Komponenten sind im log-level möglich?
Ich kenne:
- site
- lp-x
- sma
- die verschiedenen Autotypen
- die verschiedenen Wallboxtypen

Da gibt es bestimmt noch ein paar mehr?

andig, 2022-11-21:
So ganz grob per Codesuche:

- foo
- abb
- abl
- alfen
- alphatec
- amtron
- bender
- cfos
- connectiq
- dadapower
- daheim
- easee
- eebus
- etrel
- evse
- go-e
- ecb1
- salia
- heidel
- innogy
- keba
- kse
- mcc
- nrg-bt
- nrgconn
- openevse
- owbpro
- openwb
- pce
- em-eth
- ev-eth
- ev-ser
- pracht
- smaevcharger
- vestel
- wallbe
- warp
- webasto-next
- zaptec
- ocpp
- main
- mqtt
- site
- db
- semp
- bosch-bpt
- discgy
- dsmr
- modbus
- tesla
- pulse
- tq-em
- tq-em420
- fritzdect
- homematic
- lgess
- mystrom
- shelly
- tapo
- tasmota
- tplink
- cache
- http
- prometheus
- script
- ws
- sma
- push
- influx
- server
- db"
- awattar
- tibber
- etron
- audi
- carwings
- fiat
- ford
- jlr
- mercedes
- nissan
- niu
- ovms
- porsche
- citroen
- ds
- opel
- peugeot
- cupra
- seat
- s01
- enyaq
- skoda
- smart
- tronity
- volvo
- id
- vw
- kia
- hyundai

VolkerK62, 2022-11-21:
ups .... die Autos und Charger lass ich raus. Das würde den Rahmen sprengen.

Ich würde in der Doku die folgenden noch aufnehmen
- cache
- db
- influx
- mqtt

oder hast du noch welche, die wichtig wären?

PR #246 (2022-12-09): Ergänzung mögliche Komponenten bei Log-Level
fix #245
commit 2685500b (2022-12-09): Ergänzung mögliche Komponenten bei Log-Level (#246)

* Ergänzung mögliche Komponenten bei Log-Level

fix #245

* Update log.md
closed 2022-12-09 by premultiply