evcc-io/docs #264: SoC vs Soc vs soc

VolkerK62, 2023-01-15:
@naltatis 
angeregt durch https://github.com/evcc-io/evcc/issues/5787 habe ich mal die Doku (Referenz: evcc.yaml) durchgeschaut und bzgl "soc, SoC, Soc" einige änderungbedürftige Fälle gefunden.
Braucht es da jedesmal einen PR oder geht das evtl einfacher, so nach dem Motto "suchen und ersetzen"?

andig, 2023-01-15:
Suchen/ersetzen klingt gut. Eventuelle False Positives können wir vor merge immer noch korrigieren. Vielen Dank!

VolkerK62, 2023-01-15:
@andig  bei API ist mir etwas aufgefallen, bin aber nicht sicher, was richtig ist
REST API: /api/prioritysoc
MQTT API: evcc/site/prioritySoC

andig, 2023-01-15:
Aus der http.go:

	// site api
	routes := map[string]route{
		"health":        {[]string{"GET"}, "/health", healthHandler(site)},
		"state":         {[]string{"GET"}, "/state", stateHandler(cache)},
		"buffersoc":     {[]string{"POST", "OPTIONS"}, "/buffersoc/{value:[0-9.]+}", floatHandler(site.SetBufferSoc, site.GetBufferSoc)},
		"prioritysoc":   {[]string{"POST", "OPTIONS"}, "/prioritysoc/{value:[0-9.]+}", floatHandler(site.SetPrioritySoc, site.GetPrioritySoc)},
		"residualpower": {[]string{"POST", "OPTIONS"}, "/residualpower/{value:[-0-9.]+}", floatHandler(site.SetResidualPower, site.GetResidualPower)},
		"sessions":      {[]string{"GET"}, "/sessions", sessionHandler},
		"telemetry":     {[]string{"GET"}, "/settings/telemetry", boolGetHandler(telemetry.Enabled)},
		"telemetry2":    {[]string{"POST", "OPTIONS"}, "/settings/telemetry/{value:[a-z]+}", boolHandler(telemetry.Enable, telemetry.Enabled)},
		"tariff":        {[]string{"GET"}, "/tariff/{tariff:[a-z]+}", tariffHandler(site)},
	}

	for _, r := range routes {
		api.Methods(r.Methods...).Path(r.Pattern).Handler(r.HandlerFunc)
	}

	// loadpoint api
	for id, lp := range site.Loadpoints() {
		loadpoint := api.PathPrefix(fmt.Sprintf("/loadpoints/%d", id+1)).Subrouter()

		routes := map[string]route{
			"mode":          {[]string{"POST", "OPTIONS"}, "/mode/{value:[a-z]+}", chargeModeHandler(lp)},
			"targetenergy":  {[]string{"POST", "OPTIONS"}, "/targetenergy/{value:[0-9.]+}", floatHandler(pass(lp.SetTargetEnergy), lp.GetTargetEnergy)},
			"targetsoc":     {[]string{"POST", "OPTIONS"}, "/targetsoc/{value:[0-9]+}", intHandler(pass(lp.SetTargetSoc), lp.GetTargetSoc)},
			"minsoc":        {[]string{"POST", "OPTIONS"}, "/minsoc/{value:[0-9]+}", intHandler(pass(lp.SetMinSoc), lp.GetMinSoc)},
			"mincurrent":    {[]string{"POST", "OPTIONS"}, "/mincurrent/{value:[0-9.]+}", floatHandler(pass(lp.SetMinCurrent), lp.GetMinCurrent)},
			"maxcurrent":    {[]string{"POST", "OPTIONS"}, "/maxcurrent/{value:[0-9.]+}", floatHandler(pass(lp.SetMaxCurrent), lp.GetMaxCurrent)},
			"phases":        {[]string{"POST", "OPTIONS"}, "/phases/{value:[0-9]+}", phasesHandler(lp)},
			"targetcharge":  {[]string{"POST", "OPTIONS"}, "/targetcharge/{soc:[0-9]+}/{time:[0-9TZ:.-]+}", targetChargeHandler(lp)},
			"targetcharge2": {[]string{"DELETE", "OPTIONS"}, "/targetcharge", targetChargeRemoveHandler(lp)},
			"vehicle":       {[]string{"POST", "OPTIONS"}, "/vehicle/{vehicle:[1-9][0-9]*}", vehicleHandler(site, lp)},
			"vehicle2":      {[]string{"DELETE", "OPTIONS"}, "/vehicle", vehicleRemoveHandler(lp)},
			"vehicleDetect": {[]string{"PATCH", "OPTIONS"}, "/vehicle", vehicleDetectHandler(lp)},
			"remotedemand":  {[]string{"POST", "OPTIONS"}, "/remotedemand/{demand:[a-z]+}/{source::[0-9a-zA-Z_-]+}", remoteDemandHandler(lp)},
		}

Da hatte ich alles konsequent klein geschrieben. Nciht schön, aber das würde ich jetzt auch nciht ändern. Wenn Du search/replace machen willst also am besten nur noch `SoC` case sensitive suchen!

closed 2023-01-15 by andig

ammann-ruchti, 2023-01-16:
Wie wollt ihr nun den Mix aus Soc und SoC beheben? Beim Vehicle haben wir nun minSoC und targetSoC, im Messaging vehicleSoc. Wahrscheinlich muss man das so lassen, sonst werden alle aktiven evcc.yaml ungültig. Oder ihr akzeptiert beide Varianten im Server.
Oder sehe ich da etwas falsch?

VolkerK62, 2023-01-16:
die Konfig evcc.yaml ist davon nicht betroffen. 
Es geht nur um Variablen.

ammann-ruchti, 2023-01-16:
Als Anwender hast Du aber trotzdem das Problem, dass wir in der evcc.yaml beide Schreibweisen haben. Also z.B. im Vehicle targetSoC und im Messaging vehicleSoc.

VolkerK62, 2023-01-16:
Die Schriebweise in der yaml bei den Parametern ist völlig egal.
Du kannst in der yaml beim vehicle auch `targetsoc` oder `TARGETSOC` schreiben.

Lediglich wenn es um Variablen z.B. beim Messaging geht, dann ist die richtige Schreibweise wichtig.
Im Zweifel bei `/api/state` nachschauen

andig, 2023-01-16:
> Als Anwender hast Du aber trotzdem das Problem, dass wir in der evcc.yaml beide Schreibweisen haben. Also z.B. im Vehicle targetSoC und im Messaging vehicleSoc.

du hast gar kein Problem, weil an der Stelle jede Schreibweise erlaubt ist! Und wenn es hülfe könnten wir das der Einfachheit halber natürlich für die Variablen auch machen.

ammann-ruchti, 2023-01-17:
Ich habe mich bei meinen Überlegungen an Linux orientiert, wo Gross-/Kleinschreibung generell eher relevant ist, was vielleicht nicht ganz zielführend war. Wenn du die Variablen auch unabhängig von der Schrebweise machen kannst, wäre dies sicher hilfreich. Aber nur, wenn es nicht zu viel Aufwand ist (es gibt ja noch wichtigere Aufgaben). Andernfalls wäre ein deutlicher Hinweis in der Dok, dort wo die Variablen beschrieben sind, hilfreich.

premultiply, 2023-01-17:
Die Doku lebt vom mitmachen.
Für Verbesserungen einfach in der Doku editieren (unten ist ein entsprechender Link zum Doku-Repository auf jeder Seite) und daraus einen PR erzeugen.
Ist eigentlich ganz einfach. Einfach mal ausprobieren.

ammann-ruchti, 2023-01-17:
Ja, mach ich gerne, wenn mir wieder mal etwas auffällt.

PR #268 (2023-01-15): messaging: SoC -> Soc
fix #264
PR #266 (2023-01-31): site: Soc -> Soc
fix #264
commit 20fc3c18 (2023-01-31): site: Soc -> Soc

fix #264
PR #265 (2023-01-31): SoC -> Soc
fix #264