evcc-io/docs #274: Doku: modbusproxy

VolkerK62, 2023-01-25:
`modbusproxy` habe ich in der Doku nicht gefunden.
Ich denke unter dem Punkt `Modbus` würde es nicht so gut passen, sondern eher unter `Referenz/evcc.yaml` als eigener Unterpunkt.
Leider weiss ich nicht, wie ich dort einen neuen Menüpunkt einfügen kann.

andig, 2023-01-25:
Ich glaub Modbus wär gar nicht schlecht als Sonderthema- Modbus für Geräte die nur 1 Verbindung erlauben?

VolkerK62, 2023-01-25:
ich habe auch überlegt. Passt thematisch natürlich dahin, ist aber auch eine eigenes item in der yaml.
Ich bau es dann in `Modbus` ein.

VolkerK62, 2023-01-25:
ich habe es gerade mal ausprobiert. Lesen klappt, aber schreiben nicht. Kann man da nur lesen?

VolkerK62, 2023-01-25:
hmm. ok.
Wenn man mehrere Geräte abgefragen möchte, dann für jedes Gerät einen eigenen `modbusproxy` Eintrag in der yaml?  
Immer mit `port: 5200`?

andig, 2023-01-25:
Na die Ports und Zieladressen müssen schon stimmen- also jedes Ziel ein Eintrag

PR #275 (2023-01-31): add modbusproxy
fix #274
commit 49f30c1e (2023-01-31): add modbusproxy

fix #274
closed 2023-01-31 by naltatis