evcc-io/docs #290: Doku: go-e charger

VolkerK62, 2023-02-03:
Ja, vorher wurde die Ladung seitens evcc **forciert** oder unterbunden. Auch nicht immer eine gute Idee. Inkompatibel mit RFID.
Jetzt wird die Ladung seitens evcc **zugelassen** oder unterbunden.

_Originally posted by @premultiply in https://github.com/evcc-io/evcc/discussions/5990#discussioncomment-4861395_

VolkerK62, 2023-02-03:
Verstehe ich das richtig?
Wenn go-e im Eco-Modus, dann MUSS mit RFID freigeschaltet werden um die Ladung zu starten, da evcc nur zulässt und nicht forciert.
Ist der Eco-Modus ausgeschaltet, dann steuert evcc den Beginn der Ladung, auch ohne RFID -Freigabe,

premultiply, 2023-02-03:
Glaube nicht. Im Eco-Modus macht der goecharger wohl irgendeine eigene PV-Steuerung, Awattar oder sowas.
Also zwei Steuerungen für ein Gerät...

VolkerK62, 2023-02-03:
Ah, die Wirkung vom Eco-Modus war mir nicht klar. Ich lese mich dazu gerade durch verschiedene Diskussionen und Issues.
Ok. dann kann ich hier wieder zu machen. Ist ja dann ein Selbstgemachter.
Danke.

closed 2023-02-03 by VolkerK62