evcc-io/docs #317: Zählervorzeichen dokumentieren

andig, 2023-02-13:
Siehe https://github.com/evcc-io/evcc/issues/6138#issuecomment-1423785038

- Positiv: Erzeugung, Batterieentladung, Hausverbrauch, Netzbezug, Fahrzeugladung.
- Negativ: Einspeisung, Batterieladung

VolkerK62, 2023-02-17:
ist doch drin .... https://docs.evcc.io/docs/reference/configuration/meters
`evcc benutzt positive (+) Werte für eingehende Ströme (Netzbezug, PV Erzeugung, Hausbatterie-Entladung) und negative (-) Werte für ausgehende Ströme (Netzeinspeisung, PV Wechselrichter Ruhestrombedarf oder Hausbatterie-Ladung). Jeder anderweitiger Strombedarf, inklusive der Wallbox, ist immer ein positiver (+) Wert.`

premultiply, 2023-02-27:
Ja, passt schon.

closed 2023-02-27 by premultiply