evcc-io/docs #326: Awattar/Tibber Cheap gibts nicht mehr

VolkerK62, 2023-02-24:
Cheap gibts nicht mehr. Stattdessen kannst Du die Ladeplanung verwenden, siehe https://github.com/evcc-io/evcc/discussions/5639. Zur Prüfung des Tarifes:

    evcc tariff

_Originally posted by @andig in https://github.com/evcc-io/evcc/discussions/6398#discussioncomment-5092817_

VolkerK62, 2023-02-24:
@andig `cheap` kann ersatzlos gestrichen werden?

- https://docs.evcc.io/docs/reference/configuration/tariffs
- https://github.com/evcc-io/evcc/blob/master/evcc.dist.yaml

andig, 2023-02-24:
Siehe https://github.com/evcc-io/evcc/discussions/6398#discussioncomment-5098187, aktuell gibts das nicht- ggf. würde es zurück kommen. Die https://github.com/evcc-io/evcc/blob/master/evcc.dist.yaml korrigiere ich, da muss `cheap` aktuell raus. Danke für den Hinweis!

VolkerK62, 2023-03-08:
https://github.com/evcc-io/evcc/issues/6708

VolkerK62, 2023-03-20:
Parameter `smartCostLimit` in `site`.
- EUR (Tarif)
- % ??? (Grünstrom)
- gCO2

VolkerK62, 2023-03-21:
MQTT in API Doku ergänzen
https://github.com/evcc-io/evcc/pull/6971

VolkerK62, 2023-03-21:
@andig 
Wenn ich es richtig verstanden habe, dass ist es eine entweder/oder Entscheidung.
Entweder ich nutze `tariffs` (Zonen- oder variabel) oder ich nutze `planner` (z.B. GSI)  
Wenn ich einen variablen Tarif konfiguriert habe gebe ich bei `smartCostLimit` einen Preis an (analog wie früher bei cheap).
Habe ich einen fixen Tarif und z.B. GSI konfiguriert, gebe ich ein gCO2equivalent an.
Dazu noch eine Frage: auf "gruenstromindex.de" bekomme ich den %ualen Anteil grünen Stroms angezeigt (z.B. 81%).
Den kann ich hier aber nicht angeben, sondern nur gCO2. Wo kommt das her? Woher weiß ich, wieviel gut oder schlecht ist?

andig, 2023-03-21:
> Wenn ich es richtig verstanden habe, dass ist es eine entweder/oder Entscheidung.

Ne, das ist falsch.

Der Planner nutzt immer die Tarife. Man kann aber fürs Planen einen anderen (den "planner") Tarif nutzen als für den Netzbezug. Ladekosten nutzen immer "grid", der Planner nutzt "planner" oder "grid".

> sondern nur gCO2. Wo kommt das her? Woher weiß ich, wieviel gut oder schlecht ist?

Aus dem GSI Api. Weniger ist besser- irgendwann musst Du ja laden...

PR #353 (2023-03-23): add smartCostLimit to site
fix #326
commit c2ce340b (2023-03-23): add smartCostLimit to site

fix #326
PR #352 (2023-03-23): delete `cheap`
fix #326
commit 059f607d (2023-03-23): delete `cheap`

fix #326
closed 2023-03-23 by naltatis