evcc-io/docs #335: Erklärung RFC 3339

VolkerK62, 2023-02-28:
@naltatis oder @andig 
https://docs.evcc.io/docs/reference/api
`POST /api/loadpoints/<id>/target/time/<time>: target time in RFC3339 / ISO format`

Ich glaube, die wenigsten wissen, in welcher Form hier Datum und Uhrzeit einzutragen sind.
Ich gehöre dazu ;-)

Hier wäre ein Info-Feld ganz nützlich.

Wäre `2023-02-28T07:30:00` oder `2023-02-28_07:30:00` richtig?
Ist das dann UTC? Muss dann noch ein `+01:00` dahinter?
Muss das in " " gepackt werden?
Geht da auch nur Uhrzeit, ohne Datum? ist das dann täglich?

andig, 2023-02-28:
https://www.rfc-editor.org/rfc/rfc3339, keine Anführungszeichen

VolkerK62, 2023-02-28:
Die Seite hatte ich auch schon entdeckt ;-)
Sorry, dass ich nochmal nachfrage (in dem Thema bin ich ein DAU)

Also ist 28.2.2023 um 17:30 Uhr Ortszeit in Deutschland = `2023-02-28T17:30:00+01:00` (und im Sommer dann `+02:00`)

Es muss Datum & Uhrzeit sein?
Nur eine Uhrzeit geht nicht?

VolkerK62, 2023-03-01:
@andig gibst du mit bitte noch einen kurzen hint, damit ich keinen Müll in die Doku schreibe

VolkerK62, 2023-03-03:
Ich habe einfach mal try&error gemacht. Die Uhrzeit muss in UTC (`Z`) angegeben werden. Ein `+01:00` am Ende gibt immer ein "404 page not found" aus.

premultiply, 2023-03-03:
Könnte auch daran liegen dass ein + oder auch : in einer URL ggf. ein Sonderzeichen ist.

Schon vorab urlencode() probiert?

VolkerK62, 2023-03-03:
urlencode???  Du setzt wieder zu viel Wissen voraus ;-)

```
result:
2023-03-05T07%3A00%3A00%2B01%3A00
 your call:
$returnValue = urlencode('2023-03-05T07:00:00+01:00');
```

premultiply, 2023-03-03:
🤓🤖👌

VolkerK62, 2023-03-03:
Jetzt lass mich nicht verhungern.  Was sagt mir das "result" jetzt?

VolkerK62, 2023-03-03:
kommt auch wieder (erwartbar?) 404
Also muss targettime in UTC angegeben werden, da das `+` wohl das Problem ist

premultiply, 2023-03-03:
Ja ok, schade.
Einen Versuch war's aber wert und du weisst jetzt was urlencode() macht ;)

VolkerK62, 2023-03-03:
ok. Dann ist der PR richtig.

PR #339 (2023-03-03): info RFC3339
fix #335
commit bfcc36a1 (2023-03-03): info RFC3339 (#339)

fix #335
closed 2023-03-03 by premultiply