evcc-io/docs #357: OCPP stationID

VolkerK62, 2023-03-26:
Bzgl `stationid` gibt es zwei Dinge:

1. evcc braucht die nicht, solange sich nur *eine* Wallbox verbindet. Die erste Box wird einfach mit dem ersten Ocpp Loadpoint verknüpft für den ebenfalls keine `stationid` konfiguriert ist. Bei >1 Ocpp Ladepunkt geht das aber nicht mehr- wir müssen ja wissen, welche Box zu welchem Loadpoint gehört.
2. evcc verwendet `ws://<evcc>:8887/` als Url für Ocpp. Ocpp funktioniert aber so, dass die Boxen sich mit ihrer `stationid` anmelden, und zwar indem sie die anhängen; also `ws://<evcc>:8887/<stationid>`. Einige Boxen scheinen das bei Konfiguration der Url automatisch zu machen, bei einigen scheint der Anwender das anhängen zu müssen (was ziemlich verwirrend ist).

/cc @VolkerK62 bzgl. Ocpp Doku.

_Originally posted by @andig in https://github.com/evcc-io/evcc/discussions/4417#discussioncomment-5430759_

VolkerK62, 2023-03-26:
@andig das händische Anhängen der `stationid` muss dann in der Wallboxkonfiguration bei der Backend-URL erfolgen?

andig, 2023-03-26:
Ja genau, aber anscheinend eben nur manchmal- da bin ich leider raus.

VolkerK62, 2023-03-26:
Wenn man die stationid IMMER zur URL hinzufügt, dann gibt es wahrscheinlich einen Fehler, bei den Boxen die das automatisch machen?

closed 2023-04-14 by andig