evcc-io/docs #401: mqtt API: evcc/loadpoints/<id>/phase not as described

kagehisa, 2023-06-20:
While trying to create a sensor config for HASS.io I tried to setup up the sensors according to the EVCC API documentation.
In the documentation there is an API endpoint called 'evcc/loadpoints/<id>/phase', which is not present in my setup. Rather the endpoints 'evcc/loadpoints/<id>/phasesEnabled' and 'evcc/loadpoints/<id>/phasesActive' are listed according to my MQTT Explorer. 

Was there a recent API change that is not yet documented or is this an issue unrelated to the API itself and due to a configuration error on my side?

Currently running EVCC version 0.118.1

kagehisa, 2023-06-24:
Ok I changed the phases sensor definition to the following:
```yaml
    - name: evcc_phases
      unique_id: uniqueid__evcc_phases_select
      icon: mdi:backburger
      state_topic: "evcc/loadpoints/1/phasesConfigured"
      command_topic: "evcc/loadpoints/1/phases/set"
      availability_topic: "evcc/status"
      options:
        - "0"
        - "1"
        - "3"
```
The phases are set via the topic `evcc/loadpoints/1/phases/set` and read back with `evcc/loadpoints/1/phases/set`. 
Is this the way to do it, to me it is still feels weird that there is no readable entity of `evcc/loadpoints/1/phases`? If so could there be a small hint regarding this in the API description?

premultiply, 2023-06-24:
The phases are set via the topic evcc/loadpoints/1/phases/set and read back with evcc/loadpoints/1/phases

kagehisa, 2023-06-24:
Ok, then why does this Topic exists in my Setup then? Is there a condition for this Topic to be visible?

naltatis, 2023-12-21:
Outdated. https://docs.evcc.io/en/docs/reference/api/#mqtt-api hat ein Beispiel.

closed 2023-12-21 by naltatis