evcc-io/docs #457: evcc backup and restore

VolkerK62, 2023-10-14:
Hallo,

gibt es eine Anleitung wie man ein komplettes Backup vom evcc auf einem Raspberry Pi macht und anschließend auf einem neu installiertem Raspberry Pi wiederherstellen kann? Ich würde gerne alle Daten wie Ladevorgänge usw. auf das neue System übertragen.

Danke für eure Hilfe.

_Originally posted by @Peter54879 in https://github.com/evcc-io/evcc/discussions/10311_

VolkerK62, 2023-10-14:
yaml kopieren:  `sudo cp /etc/evcc.yaml /home/pi/evcc.yaml.bak`
db kopieren: `sudo cp /var/lib/evcc/evcc.db /home/pi/evcc.db.bak

Peter54879, 2023-10-14:
Das Backup hat funktioniert.
Leider bekomme ich die .db nicht auf den neuen Raspberry Pi überschrieben.

closed 2023-10-29 by naltatis

Marty56, 2024-03-05:
Ein Kopie eine laufenden SQL Datenbank ist u.U. korrupt.
Normalerweise nimmt man dazu ein Backup Mechanismus der DB

Aitschcux, 2024-05-01:
Hallo!
Wie könnte ich denn eine evcc.db.bak, welche ich zuvor von meinem alten Raspberry Pi auf meinen Mac gesichert habe, z. B. per ssh auf meinen neu installierten Raspberry Pi zurück kopieren? Hier würde ich mich freuen, wenn die Doku bzgl. des o. g. Beispiels noch ergänzt werden könnte. Besten Dank im Voraus und vielen Dank für Eure Hilfe!

VolkerK62, 2024-05-02:
Im Prinzip, wie in der Doku beschrieben, nur Speicherort/name und Zielort/name tauschen

hin:  `sudo cp /var/lib/evcc/evcc.db /home/pi/evcc.db.bak`

zurück:  `sudo cp /home/pi/evcc.db.bak /var/lib/evcc/evcc.db`

PR #458 (2023-10-29): backup and restore
fix #457