evcc-io/docs #484: Remove obsolete settings

andig, 2023-12-25:
Weil du als Entwickler irgendwann die obsolete Konfiguration nicht mehr im Code berücksichtigen willst. Irgendwann kommt der Punkt, wo aus einer Warnung ein Fehler werden muss. Wenn nicht jetzt, dann in drei Monaten, und dann meckern die Leute genau so. Ich bin auch drüber gestolpert und habe mich drüber geärgert, weil ich die Ausgabe im Logfile (siehe [https://github.com/evcc-io/evcc/discussions/10978#discussioncomment-7933198](url)) übersehen habe.

@andig wenn mir eine Manöverkritik erlaubt ist: In der Dokumentation ist nicht erkennbar, dass diese Parameter _deprecated_ sind bzw. nun gar nicht mehr verwendet werden sollen. Beispiel siehe [https://docs.evcc.io/docs/reference/configuration/loadpoints#resetondisconnect](url). Ihr hättet euch etwas möglicherweise übellaunige Kommentare ersparen können, wenn die Parameter dort schon vor längerer Zeit als _deprecated_ gekennzeichnet worden wären. Ansonsten: Danke für EVCC!

_Originally posted by @jph76 in https://github.com/evcc-io/evcc/discussions/10978#discussioncomment-7943551_

VolkerK62, 2024-01-04:
site:
- `smartCostLimit` ist noch offen, PR besteht
- `battery socs` sind noch offen, PR besteht

loadpoints:
- `resetOnDisconnect` ist gelöscht
- `mode` ist entsprechend angepasst

vehicles:
- `minsoc` ist gelöscht
- `targetsoc` ist gelöscht

andig, 2024-01-04:
Dann machen wir hier schonmal zu- tausend Dank!

closed 2024-01-04 by andig