evcc-io/docs #520: Add usage=aux to meters page

naltatis, 2024-01-28:
Add new aux type to https://docs.evcc.io/docs/devices/meters

premultiply, 2024-01-28:
Hier haben wir noch generell ein bisschen das kosmetische Problem dass Geräte abhängig von ihrer usage (teilweise auch mehrfach) in brands.json und damit auf der Webseite bei den Herstellern der Gerätekategorien auftauchen.

Das führt machmal zu lustigen Effekten wie z. B. dass ein Modbus-Zähler-Hersteller (der ja z. B. auch ein PV-Erzeugungszähler sein kann) dort als Wechselrichterhersteller auftaucht. Gleiches gilt ähnlich für `evcc configure`.

andig, 2024-02-06:
Es wird langsam dringender- die ersten Configs mit "undefined" tauchen auf: 
[image]

naltatis, 2024-02-06:
Doku ist angepasst. Bei `configure` müsste jemand von euch bitte mal reinschauen. Außerdem ist mir der hier aufgefallen:

[image]

Finde den Beschreibungstext etwas verwirrend. Gerade weil der Inhalt nicht zu den Metadaten (`usage`) passt.

closed 2024-02-06 by naltatis