evcc-io/docs #537: Enhance devices/charges.mdx docu for E3/DC Easy Connect wallbox

docolli, 2024-03-04:
From the discussions about evcc I have seen, that users still are not aware about how to setup evcc with E3/DC Easy Connect wallbox and the home power station correctly by using the modbusproxy and connecting the home power station via evcc to the wallbox. The solution "you have to delete the wallbox in your home power system to get it working with evcc" is no longer valid!

Since I have got a lot of experience meanwhile with my E3/DC system and also evcc, I started this weekend to improve the E3/DC Easy Connect section of chargers.mdx to help users to setup their E3/DC wallbox and system with evcc. I already have setup a local development system, but since I am a complete newbie to git/github, I am asking you now, how should I present you my changes for discussion/release?

Should I just do my changes locally and then start a pull request? Or is there another way to do so? What is the desired proceedure?

naltatis, 2024-03-04:
Device specific configuration is located in evcc core project in the individual template files. For e3dc devices it's this one:
https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/e3dc.yaml

Use the `requirements.description.[en|de]:` field in the template file to describe the tasks a user has to perform to get it working.

See Kostal Plenticore as an example.
The text in this template
https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/kostal-plenticore.yaml#L13-L18
is rendered in docs like this
https://docs.evcc.io/docs/devices/meters#plenticore-hybrid

naltatis, 2024-03-04:
> Or should I add a "Wallbox" Tab into this core project description?
Not sure, what you mean with this.

You can also adjust the existing text for the E3DC wallbox.
https://docs.evcc.io/docs/devices/chargers#e3dc-easy-connect

naltatis, 2024-03-04:
Constraint: Just saw that the E3DC wallbox uses a generic template `phoenix-ev-eth`. Changes to this template would affect all other chargers using the same template. Currently we dont have any way around this. But you docs should be placed either at the inverter or wallbox template. We don't have (and want) other locations for documentation of specific szenarios of devices.

docolli, 2024-03-04:
> > Or should I add a "Wallbox" Tab into this core project description?
> > Not sure, what you mean with this.
> 
> You can also adjust the existing text for the E3DC wallbox. https://docs.evcc.io/docs/devices/chargers#e3dc-easy-connect

That is the location, where I started yesterday to add my docs.

**To give you some more infos to be able to help me more specific:**
The E3/DC Wallbox should no longer simply be added with its IP address to evcc, since this does not work, when the wallbox is also added to the home power station. This leads to conflicts in wallbox control. 
Deleting the wallbox in the home power station is one solution. However, there are some nice benefits, when the home power station still knows the power values of the wallbox, like correct calculation of house power (without wallbox power) and E3DC statistics in the web portal. Also the internal SOC limit until the battery of the home power station supports car loading can be used. Below this limit the home power station can deliver dynamically what the house needs, so the wallbox does not quickly drain the house battery.

To do so, the evcc modbusproxy must be set up with the IP address of the wallbox and `readonly=true` option. Then also the wallbox must be deleted from home power station and added again, but now with the IP address of evcc.

All this information (maybe with some screenshots from E3/DC home power station) I want to add to the docu to help other E3/DC Easy Connect Wallbox users.

naltatis, 2024-03-04:
> PS: With "Tab" I mean this one, that some devices have in the description

Thats not an actual charger. It's just a "meter" that ready charger related values. No controls, just reading.

To your specific case. This sounds like quite a few steps and information. I don't think we already have something like this. Another option I could propose is using the community wiki and describing your setup there https://github.com/evcc-io/evcc/wiki
You can describe all your steps there. I also think images are possible. Its somewhat findable for others users and we can then figure out a ways to integrate your content, and maybe related content like this into the official docs page.

docolli, 2024-03-04:
Good idea! I will post my experience in this area.
Since I cannot help with active coding, at least I want to support this wunderful project with help in docs.