evcc-io/docs #548: Sungrow Hybrid Inverter and Winet-S Dongle hint

martinknechtel, 2024-03-24:
@premultiply Mir scheint das kommt von deinem Commit https://github.com/evcc-io/docs/commit/8161338985ac03b0a06b7c6e44c9de43aa8005fc und ergab die Zeile https://github.com/evcc-io/docs/blame/2f668ca98e6d04b5c22362de54c9c3b3ec06bc6f/templates/release/de/meter/sungrow-hybrid_0.yaml#L6
deshalb meine Frage an dich: Worauf stützt sich die Befürchtung die Sungrow Hybrid Converter würden mit dem Winet-S Dongle nicht korrekt funktionieren. Also ich benutze das und es funktioniert eigentlich problemlos. Gibt es einen Link den ich hier zur Erklärung unterbringen könnte?

premultiply, 2024-03-24:
Das Dongle hat diverse Fehler in der Modbus-Implementierung und liefert z.B. keine Vorzeichen für die Batterie so dass die Schnittstelle nur auf den ersten Blick richtig funktioniert in der Praxis aber unbrauchbar ist.

closed 2024-03-24 by premultiply

martinknechtel, 2024-03-25:
Ach deshalb wird bei der Batterie nur der % SoC gezeigt, nicht aber der Ladestrom? Gibt es da irgendeinen Link den ich dranpacken kann, damit nicht der nächste Leser drüber stolpert?