evcc-io/docs #550: Add Tariffs section

andig, 2024-04-02:
Refs https://github.com/evcc-io/evcc/pull/10042

naltatis, 2024-04-02:
@andig Worum gehts hier? Die im PR neu dazukommenden API Calls dokumentieren? Tariff Sections haben wir ja bereits:

User: https://docs.evcc.io/docs/features/dynamic-prices
Reference: https://docs.evcc.io/docs/reference/configuration/tariffs/

andig, 2024-04-03:
Ich hatte die Tarife in https://docs.evcc.io/docs/reference/configuration/tariffs/ gar nicht gefunden. 

Vorschlag: dort rauswerfen und auf Basis Templates unter "Geräte" oder auf gleicher Ebene ergänzen. Daten aus den Templates. Dort wo nicht über Templates konfigurierbar sollten wir Pseudo-Templates anlegen die zumindest die Beschreibung enthalten. Format tbd (Markdown?).

naltatis, 2024-05-08:
Fixed with https://github.com/evcc-io/docs/pull/562

closed 2024-05-08 by naltatis