evcc-io/docs #560: Document: Loadpoint: add welcomecharge feature

andig, 2024-04-25:
We need to document the new feature introduced in this PR: https://github.com/evcc-io/evcc/pull/13534

andig, 2024-05-10:
On hold

andig, 2024-08-11:
@naltatis Anladen selbst ist nicht dokumentiert, oder?

naltatis, 2024-08-11:
An den Fahrzeugen die es unterstützen/benötigen schon. An was hattest du denn gedacht?

andig, 2024-08-11:
Ich dachte es sollte auch selbst in der Doku erwähnt werden, Sinn & Zweck.

naltatis, 2024-08-11:
Sag mal, was du damit meinst. Das ist ja eigentlich kein tolles Nutzerfeature, sondern eher ein Workaround, das nur für bestimmte Fahrzeuge notwendig ist. Daher find ich den Ort es genau an diesen Fahrzeugen zu zeigen ganz gut.

Deine Idee wäre bspw. auf der Funktionen > Fahrzeuge Seite da einen Absatz (analog Klima) zu zu schreiben?

TobiasHuber1980, 2024-08-11:
> ...Daher find ich den Ort es genau an diesen Fahrzeugen zu zeigen ganz gut.
> 
> 
> 
> Deine Idee wäre bspw. auf der Funktionen > Fahrzeuge Seite da einen Absatz (analog Klima) zu zu schreiben?

Es gibt auch wallbox(en) z.B. KEBA die das Feature benötigen (unabhängig vom Fahrzeug)