evcc-io/docs #568: Meter missing powers

andig, 2024-06-16:
this is working well! thanks!
I mean the generic template here: [https://docs.evcc.io/docs/devices/meters#](url)

_Originally posted by @pdonn in https://github.com/evcc-io/evcc/issues/13867#issuecomment-2171804761_

VolkerK62, 2024-06-17:
zusätzlich zu `powers` fehlt auch noch `voltages`.

Ist das ein Template (hab keins gefunden), oder kann man das direkt in der Doku ändern?

premultiply, 2024-06-17:
Müsste man direkt in der Doku ändern können, oder @naltatis ?

andig, 2024-06-17:
Dann sollten wir auch gleich dazu schreiben wofür die jeweils verwendet werden.

- Spannungen: Anschlusserkennung 1p3p
- Ströme: Erkennung aktive Phasen
- Leistungen: Vorzeichenerkennung für Ströme falls diese vorzeichenlos sind

PR #569 (2024-06-19): Doku: custom meter, add powers and voltages
fix #568
closed 2024-06-19 by naltatis