evcc-io/docs #622: Anzeige nicht verfügbare Funktion an Geräten

StefanSchoof, 2024-09-16:
In der Doku gibt es die verschiedenen Eigenschaften nach denen gefiltert werden kann.
https://docs.evcc.io/docs/devices/chargers#nach-eigenschaften-filtern

Bei den Geräten wird nur anzeigt, welches Feature die Geräte haben. Wenn ein neuer Nutzer sein Gerät sucht ist ihn die Funktion nicht immer bewusst. Ich glaube es wäre gut, wenn an den Geräten auch angezeigt wird, welche Eigenschaften nicht vorhanden sind. Also z.B. das eine Wallbox nicht 1p3p fähig ist.

Sonst führt auch zu Dingen, dass in der Beschreibung "Netzladung ist generell nicht verfügbar." erfasst wird, wie z.B. https://github.com/evcc-io/evcc/pull/16145/files Dies ist meiner Meinung nach eine Info die besser nur bei Geräten erfasst wird, die es können. Aber ich kann verstehen, dass Nutzer gerne einfach sehen wollen, was nicht mit dem Gerät geht.

naltatis, 2024-09-23:
Ich bin hier etwas hin und hergerissen. Verstehe den Punkt, würde aber ungern an jedem Eintrag eine stetig wachsende Liste an Funktionen haben. Bei deinem konkreten Beispiel gibt es, finde ich, auch noch mal zwei Ebenen. Bei dem Gerät wurde geprüft, ob Netzladen möglich ist und dann bewusst gesagt. Ne, technisch nicht möglich. Das ist ja noch mal eine andere Aussage, als wenn noch Niemand danach gefragt und geschaut hat. Also der Unterschied zwischen "Gerät kann das nicht" und "evcc hat es nicht implementiert".

StefanSchoof, 2024-09-25:
Ich verstehe, dass du nicht zu viele Dinge anzeigt haben willst. Aber ich denke für eine gute User Experience ist es wichtig, was sie von ihrer Hardware erwarten können. Wenn sie gehört haben, dass evcc Phasenumschaltung kann, sollte man ihnen so früh wie möglich sagen, dass evcc das nicht mit ihrer Hardware kann.

StefanSchoof, 2024-09-30:
Hier führt es dazu das die Informationen jetzt in der Beschreibung steht:

> Die Nutzung der Phasenumschaltung ist in evcc momentan noch nicht möglich.

https://github.com/evcc-io/evcc/blob/50bbeeafcba8bc8daab082e45281def3251f03f1/templates/definition/charger/ghost.yaml#L16