evcc-io/docs #650: Available measurements

andig, 2024-10-23:
At time of writing, ignoring setters:

## charger

- **status**
- power
- energy (total import)
- chargedenergy (not implemented for `custom` charger)
- currents
- voltages
- powers (not implemented for `custom` charger)
- soc (vehicle or integrated device)
- phases1p3p
- getmaxcurrent (not implemented for `custom` charger)
- chargeduration (not implemented for `custom` charger)

## meter

- **power**
- energy
- currents
- voltages (used for connected phases)
- powers (used for signed currents)
- soc (battery)
- capacity (battery)
- maxacpower (inverter)

## vehicle

- **soc** (vehicle)
- status
- range
- odometer
- climater (used for active detection)
- getmaxcurrent (only used with TWC)
- finishtime

## tariff

- price
- forecast

andig, 2024-10-27:
Es ging um Doku für die custom Geräte, v.a. aber auch darum zu dokumentieren, welche Messwerte überhaupt für welchen Zweck sinnvoll sind. In letzter zeit gibts den Trend, alle möglichen Dinge einzubauen die aber überhaupt nie verwendet werden...

andig, 2024-10-27:
Anderes Thema wäre die aktuelle custom Doku raus zu ziehen und aufzubohren ;)

naltatis, 2024-12-10:
will be covered by https://github.com/evcc-io/docs/issues/678

closed 2024-12-10 by naltatis