evcc-io/docs #732: tariff settings time dependent - one hour off?

mheckner, 2025-02-13:
Hello,

I have Intelligent Octopus Go, that offers a reduced price (16.09 ct per kwh) between 0:00-5:00 in the morning, the rest of the day are 28.49 ct per kwh. I also configured a price limit of <= 16.5 ct per kwh in the UI for charging in PV mode when the energy price is low.

My settings in `evcc.yaml` are as follows:

```
tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.2849
    zones:
      - days: Mo-So
        hours: 0-5
        price: 0.1609
  feedin:
    type: fixed
    price: 0.095
```
Running evcc tariff reflects the correct pricing as intended:

```
2025-02-18 22:00:00    2025-02-18 23:00:00    0.285
2025-02-18 23:00:00    2025-02-19 00:00:00    0.285
2025-02-19 00:00:00    2025-02-19 01:00:00    0.161
2025-02-19 01:00:00    2025-02-19 02:00:00    0.161
2025-02-19 02:00:00    2025-02-19 03:00:00    0.161
2025-02-19 03:00:00    2025-02-19 04:00:00    0.161
2025-02-19 04:00:00    2025-02-19 05:00:00    0.161
2025-02-19 05:00:00    2025-02-19 06:00:00    0.285
2025-02-19 06:00:00    2025-02-19 07:00:00    0.285
2025-02-19 07:00:00    2025-02-19 08:00:00    0.285
```
However, the UI shows that the car will charge at 01:00 ("<= 16,5 ct morgen 01:00"), whereas I expected that this would say 0:00:

[image]

The settings in the UI also seem to reflect this shift of one hour (I would expect hours 0, 1, 2, 3, 4 to be green, not 1,2,3,4,5):

[image]

Am I doing something wrong? Any hints appreciated!

naltatis, 2025-02-14:
Ok, looks like "problem solved".

naltatis, 2025-02-14:
@mheckner btw. please post issues like this in the main repo https://github.com/evcc-io/evcc. This is the docs repository.

closed 2025-02-14 by naltatis