evcc-io/docs #823: OpenAPI doesn't validate

andig, 2025-06-21:
Using https://docs.evcc.io/assets/files/rest-api-64d953eef4da0bc1e1c2a1d274f2ea57.yaml with https://oas-validation.com I'm getting erros:

```
Context: Path: /loadpoints/{id}/plan/static/preview/{type}/{goal}/{timestamp}, Method: GET, Parameter: goal, Example: socAvailable
Example:
80
Error: 80 is valid under each of {'$ref': '#/components/parameters/power'}, {'$ref': '#/components/parameters/soc'} Failed validating 'oneOf' in schema: {'oneOf': [{'$ref': '#/components/parameters/soc'}, {'$ref': '#/components/parameters/power'}]} On instance: 80
Context: Path: /loadpoints/{id}/plan/static/preview/{type}/{goal}/{timestamp}, Method: GET, Parameter: goal, Example: socNotAvailable
Example:
2500
Error: 2500 is valid under each of {'$ref': '#/components/parameters/power'}, {'$ref': '#/components/parameters/soc'} Failed validating 'oneOf' in schema: {'oneOf': [{'$ref': '#/components/parameters/soc'}, {'$ref': '#/components/parameters/power'}]} On instance: 2500
Context: Schema: HourMinuteTime
Example:
HH:MM
Error: Unrecognized format 'HH:MM'
```

/cc @Maschga

Maschga, 2025-06-21:
Ich sehe die beiden ersten Fehler als Hinweis, dass sich die beiden Beispiele `80` und `2500` nicht auf ein einzelnes Schema eingrenzen lassen, sondern sowohl als `soc` als auch als `power` validiert werden können.
Das ist aber meiner Meinung nach kein Fehler.

Für den dritten Fehler, siehe [`format` is an open value](https://swagger.io/docs/specification/v3_0/data-models/data-types/#string-formats). Das ist definitiv in Ordnung.

andig, 2025-06-21:
Leider sieht das Tooling das anders und weigert sich die Spec weiter zu verarbeiten. Inhaltlich kann ich das nicht kommentieren :(

andig, 2025-06-23:
Seems different now. https://oas-validation.com passes, but https://jedisct1.github.io/openapi-mcp/ doesn't:

```
openapi-mcp validate rest-api.yaml
Validation failed: OPENAPI LOADING ERROR: Spec parsing failed

FILE: rest-api.yaml

ORIGINAL ERROR:
OPENAPI LOADING ERROR: Spec parsing failed

ORIGINAL ERROR:
bad data in "#/components/parameters/soc" (expecting ref to schema object)
```

Maschga, 2025-06-23:
Gibt es dazu eine Zeiöenangabe? Oder ein Online Tool, in dem ich die Datei testen kann, bis keine Fehler mehr aufkommen?

Maschga, 2025-06-23:
Ich habe `openapi-mcp` zum Laufen bekommen.
Werde die Hinweise und Fehler fixen und einen PR für das Docs Repo erstellen.

andig, 2025-06-23:
Top- soll ich das validator Issue dann schliessen? Könntest du die yaml hier posten?

Maschga, 2025-06-23:
Nein, das kannst du offen lassen.
Ich würde die Yaml in Master mergen wollen.

Maschga, 2025-06-23:
@andig Reicht es aus, wenn `openapi-mcp validate` keinen Fehler zurückgibt, auch wenn https://oas-validation.com failed?

closed 2025-06-26 by naltatis