evcc-io/docs #851: EEBUS: SHIPID vs. SKI

CiNcH83, 2025-07-23:
Beim Thema EEBUS Konfiguration scheint die Doku etwas unscharf:

https://github.com/evcc-io/docs/blob/547af3a272eac3f3b9ed965e6f6d6fa1a77d77b3/docs/reference/configuration/eebus.md?plain=1#L96-L102

Was genau soll da nun hin? SHIPID oder SKI? Auch das Beispiel ist irreführend. Oder ist das eine gültige SHIPID?

DerAndereAndi, 2025-07-23:
>  Hiermit kann die zu verwendende Geräte-ID (SKI) definiert werden. 

Der Text ist falsch. Eher vielleicht so etwas:

> Hiermit kann die zu verwendete Geräte-ID (SHIP-ID) definiert werden. Dies sollte nur für Entwicklungszwecke notwendig sein.

Die SKI wird aus dem Zertifikat automatisch generiert und von der Gegenstelle auch u.U. gegen das Zertifikat (Public-Key) geprüft.

Die SKI ist normalerweise NICHT Teil der SHIP-ID, aber bei evcc wird dies mangels Alternativen gemacht.

DerAndereAndi, 2025-07-23:
Gemini hat keine Ahnung. Es gibt keine Vorgabe außer dass es ein String ist, der sich nicht verändern darf.

CiNcH83, 2025-07-23:
> der sich nicht verändern darf.

Verwendet evcc dafür `/dev/machine-id`? Wenn man evcc auf ein anderes Gerät umzieht, ändert sich diese wahrscheinlich? Evtl. dann doch besser, eine SHIPID in der evcc.yaml einzutragen, weil man sonst die Assoziierungen neu machen muss?

CiNcH83, 2025-07-23:
OK. Wenn ich früher plant und Zertifikat in der evcc.yaml hatte, konnte ich das einfach auf ein neues Gerät migrieren. Wenn ich die machine-id genutzt und nur das Zertifikat migriert habe, muss ich die Assoziierung trotzdem wegen der geänderten SHIPID (basierend auf der neuen machine-id) neu machen, oder?

Wie soll das in Zukunft ohne evcc.yaml aussehen? plantid und Zertifikat sind dann wohl in der DB, welche neuerdings gesichert und wiederhergestellt werden kann? D.h. es ist besser, die machine-id nicht in den Docker-Container zu mappen und eine plantid zu nutzen? Evtl. macht es Sinn, die Dokumentation dahingehend anzupassen?

DerAndereAndi, 2025-07-24:
Oder man koppelt einfach neu. Das wird nur mit einem SMGW/Steuerbox dann ein Problem.

Wie das in Zukunft aussehen soll, weiss ich nicht. Bin ich nicht beteiligt.

/cc @andig @naltatis

CiNcH83, 2025-07-24:
Also ich würde dann eher auf die plantid statt machine-id gehen.

@naltatis Ich gehe davon aus, dass die evcc.yaml irgendwann verschwinden wird? Wird die plantid dann samt Zertifikat in der DB gespeichert? Wird bei der Migration die plantid aus der evcc.yaml in die DB übertragen?

CiNcH83, 2025-07-24:
Ich war der Auffassung, dass die Migration von evcc.yaml auf DB nicht in allzu weiter Zukunft liegt. #20190 scheint ja schon eine ID zu generieren und in der DB zu speichern, sofern keine andere vorhanden ist. Einen Migrationsprozess gibt es ja auch bereits? Ich hätte erwartet, dass da die plantid in die DB übertragen wird? In der [runMigrate](https://github.com/evcc-io/evcc/blob/ba29e4c16ec50eb436fbdd34266c9f35c7755f6c/cmd/migrate.go#L24) kann ich das aber nicht finden.

closed 2025-10-24 by naltatis