evcc-io/docs #892: Document: Add heatpump dimming

andig, 2025-09-29:
We need to document the new feature introduced in this PR: https://github.com/evcc-io/evcc/pull/23925

RTTTC, 2025-10-24:
`sg-ready dim` via shelly relay is needed.
Please provide code sample to be used with UI if dedicated template is not planned

naltatis, 2025-10-24:
basic structure for `sgready` is available here (config ui). but we need a concrete example for a e.g. shelly with 2 relays setup:

https://github.com/evcc-io/evcc/blob/master/assets/js/components/Config/defaultYaml/sgready.yaml

andig, 2025-10-24:
Fine. Somebody with Shelly will need to do one and test it. I don't see any problem doing so and have not received any question. About same status as 3 months ago...

andig, 2025-10-24:
I don't understand. This config doesn't do anything useful.